Hypfer/Valetudo #153: Firmware build failling

d1slact0r, 2019-04-07:
When following the build instructions I get the following error:
```
/mnt/firmware/rockrobo/image/fw.0Jv7wa/sounds /mnt/firmware/rockrobo/image
/mnt/firmware/rockrobo/image
Decrypt firmware
Unpack firmware
/mnt/firmware/rockrobo/image/fw.0Jv7wa/fw /mnt/firmware/rockrobo/image
/mnt/firmware/rockrobo/image
mount: /mnt/firmware/rockrobo/image/fw.0Jv7wa/image: can't read superblock on /dev/loop7.
Replace ssh host keys
../dustcloud/devices/xiaomi.vacuum/firmwarebuilder/imagebuilder.sh: line 343: /mnt/firmware/rockrobo/image/fw.0Jv7wa/image/etc/ssh/ssh_host_rsa_key: No such file or directory
../dustcloud/devices/xiaomi.vacuum/firmwarebuilder/imagebuilder.sh: line 344: /mnt/firmware/rockrobo/image/fw.0Jv7wa/image/etc/ssh/ssh_host_rsa_key.pub: No such file or directory
../dustcloud/devices/xiaomi.vacuum/firmwarebuilder/imagebuilder.sh: line 345: /mnt/firmware/rockrobo/image/fw.0Jv7wa/image/etc/ssh/ssh_host_dsa_key: No such file or directory
../dustcloud/devices/xiaomi.vacuum/firmwarebuilder/imagebuilder.sh: line 346: /mnt/firmware/rockrobo/image/fw.0Jv7wa/image/etc/ssh/ssh_host_dsa_key.pub: No such file or directory
../dustcloud/devices/xiaomi.vacuum/firmwarebuilder/imagebuilder.sh: line 347: /mnt/firmware/rockrobo/image/fw.0Jv7wa/image/etc/ssh/ssh_host_ecdsa_key: No such file or directory
../dustcloud/devices/xiaomi.vacuum/firmwarebuilder/imagebuilder.sh: line 348: /mnt/firmware/rockrobo/image/fw.0Jv7wa/image/etc/ssh/ssh_host_ecdsa_key.pub: No such file or directory
../dustcloud/devices/xiaomi.vacuum/firmwarebuilder/imagebuilder.sh: line 349: /mnt/firmware/rockrobo/image/fw.0Jv7wa/image/etc/ssh/ssh_host_ed25519_key: No such file or directory
../dustcloud/devices/xiaomi.vacuum/firmwarebuilder/imagebuilder.sh: line 350: /mnt/firmware/rockrobo/image/fw.0Jv7wa/image/etc/ssh/ssh_host_ed25519_key.pub: No such file or directory
Disable SSH firewall rule
sed: can't read /mnt/firmware/rockrobo/image/fw.0Jv7wa/image/opt/rockrobo/watchdog/rrwatchdoge.conf: No such file or directory
Add SSH authorized_keys
mkdir: cannot create directory ‘/mnt/firmware/rockrobo/image/fw.0Jv7wa/image/root/.ssh’: No such file or directory
chmod: cannot access '/mnt/firmware/rockrobo/image/fw.0Jv7wa/image/root/.ssh': No such file or directory
../dustcloud/devices/xiaomi.vacuum/firmwarebuilder/imagebuilder.sh: line 365: /mnt/firmware/rockrobo/image/fw.0Jv7wa/image/root/.ssh/authorized_keys: No such file or directory
chmod: cannot access '/mnt/firmware/rockrobo/image/fw.0Jv7wa/image/root/.ssh/authorized_keys': No such file or directory
reconfiguring network traffic to xiaomi
../dustcloud/devices/xiaomi.vacuum/firmwarebuilder/imagebuilder.sh: line 373: /mnt/firmware/rockrobo/image/fw.0Jv7wa/image/etc/hosts: No such file or directory
../dustcloud/devices/xiaomi.vacuum/firmwarebuilder/imagebuilder.sh: line 374: /mnt/firmware/rockrobo/image/fw.0Jv7wa/image/etc/hosts: No such file or directory
replacing adbd
cp: cannot stat '/mnt/firmware/rockrobo/image/fw.0Jv7wa/image/usr/bin/adbd': No such file or directory
install: cannot create regular file '/mnt/firmware/rockrobo/image/fw.0Jv7wa/image/usr/bin/adbd': No such file or directory
Installing valetudo
install: cannot create regular file '/mnt/firmware/rockrobo/image/fw.0Jv7wa/image/usr/local/bin/valetudo': No such file or directory
install: cannot create regular file '/mnt/firmware/rockrobo/image/fw.0Jv7wa/image/etc/init/valetudo.conf': No such file or directory
../dustcloud/devices/xiaomi.vacuum/firmwarebuilder/imagebuilder.sh: line 527: /mnt/firmware/rockrobo/image/fw.0Jv7wa/image/etc/hosts: No such file or directory
sed: can't read /mnt/firmware/rockrobo/image/fw.0Jv7wa/image/etc/rc.local: No such file or directory
../dustcloud/devices/xiaomi.vacuum/firmwarebuilder/imagebuilder.sh: line 530: /mnt/firmware/rockrobo/image/fw.0Jv7wa/image/etc/rc.local: No such file or directory
../dustcloud/devices/xiaomi.vacuum/firmwarebuilder/imagebuilder.sh: line 531: /mnt/firmware/rockrobo/image/fw.0Jv7wa/image/etc/rc.local: No such file or directory
../dustcloud/devices/xiaomi.vacuum/firmwarebuilder/imagebuilder.sh: line 532: /mnt/firmware/rockrobo/image/fw.0Jv7wa/image/etc/rc.local: No such file or directory
sed: can't read /mnt/firmware/rockrobo/image/fw.0Jv7wa/image/opt/rockrobo/rrlog/rrlog.conf: No such file or directory
sed: can't read /mnt/firmware/rockrobo/image/fw.0Jv7wa/image/opt/rockrobo/rrlog/rrlogmt.conf: No such file or directory
sed: can't read /mnt/firmware/rockrobo/image/fw.0Jv7wa/image/opt/rockrobo/rrlog/rrlog.conf: No such file or directory
sed: can't read /mnt/firmware/rockrobo/image/fw.0Jv7wa/image/opt/rockrobo/rrlog/rrlogmt.conf: No such file or directory
sed: can't read /mnt/firmware/rockrobo/image/fw.0Jv7wa/image/opt/rockrobo/watchdog/ProcessList.conf: No such file or directory
sed: can't read /mnt/firmware/rockrobo/image/fw.0Jv7wa/image/opt/rockrobo/rrlog/tar_extra_file.sh: No such file or directory
sed: can't read /mnt/firmware/rockrobo/image/fw.0Jv7wa/image/opt/rockrobo/rrlog/misc.sh: No such file or directory
sed: can't read /mnt/firmware/rockrobo/image/fw.0Jv7wa/image/opt/rockrobo/rrlog/toprotation.sh: No such file or directory
sed: can't read /mnt/firmware/rockrobo/image/fw.0Jv7wa/image/opt/rockrobo/rrlog/topstop.sh: No such file or directory
../dustcloud/devices/xiaomi.vacuum/firmwarebuilder/imagebuilder.sh: line 557: /mnt/firmware/rockrobo/image/fw.0Jv7wa/image/opt/rockrobo/watchdog/ntpserver.conf: No such file or directory
../dustcloud/devices/xiaomi.vacuum/firmwarebuilder/imagebuilder.sh: line 561: /mnt/firmware/rockrobo/image/fw.0Jv7wa/image/opt/rockrobo/watchdog/ntpserver.conf: No such file or directory
../dustcloud/devices/xiaomi.vacuum/firmwarebuilder/imagebuilder.sh: line 562: /mnt/firmware/rockrobo/image/fw.0Jv7wa/image/opt/rockrobo/watchdog/ntpserver.conf: No such file or directory
../dustcloud/devices/xiaomi.vacuum/firmwarebuilder/imagebuilder.sh: line 564: /mnt/firmware/rockrobo/image/fw.0Jv7wa/image/etc/timezone: No such file or directory
umount: /mnt/firmware/rockrobo/image/fw.0Jv7wa/image: not mounted.
waiting for unmount...
```

Hypfer, 2019-04-07:
I'd say broken or incomplete fw download.
What exactly did you download from where and what is its md5 sum?

Also what do you mean by `(please note that the firmware version on the wiki page is not correct, please update that)``?

d1slact0r, 2019-04-07:
@Hypfer thanks for the quick reply, start by the last question. here: https://github.com/Hypfer/Valetudo/wiki/Installation-Instructions#creating-the-firmware-image
although here: https://github.com/Hypfer/Valetudo/wiki/Installation-Instructions#creating-the-firmware-image version downloaded is: v11_001748.fullos

How do i check the MD5? I tried redownloading but still the same issue, will use another url to download it.

Hypfer, 2019-04-07:
Ah yes. I will change the wiki to make clear that the user has to replace the path in the command with the name of the firmware he downloaded.

md5sum should be installed on basically every linux system so you just need to run `md5sum ./path/to/firmwareimage.pkg`

d1slact0r, 2019-04-07:
Thanks Hypfer, I used x and the md5 is: 4880d2b189d8af567ae96d2073579021
`wget https://cdn.cnbj2.fds.api.mi-img.com/rubys/updpkg/v11_001748.fullos.pkg

Hypfer, 2019-04-07:
`4880d2b189d8af567ae96d2073579021` is also the md5 sum of the image I have downloaded and running.
So it's not a corrupt firmware image. Still, mounting the image to edit its content fails `can't read superblock on /dev/loop7`
But why? What are you running this on?

d1slact0r, 2019-04-07:
Running it on MacOS virtualbox with ubuntu latest version 18.04

Hypfer, 2019-04-07:
Googling `can't read superblock on /dev/loop`  brings up this Bug in Virtualbox https://www.virtualbox.org/ticket/17360

Could it be related?

d1slact0r, 2019-04-07:
Nice find, that did it :) moved the stuff from the shared folder into the virtualbox itself and now it is working! Thanks :)

closed 2019-04-07 by d1slact0r

balu-, 2019-04-23:
thanks for that hint, helped me a lot!