Hypfer/Valetudo #156: Fall Back Image ?

thundergreen, 2019-04-07:
Dear all,

i faced a very interessting issue. I compiled built and transferred new firmware to my robo. Once flashing process was over i saw valetusoversion and Frirmware version in correct manner. After reboot i got like a fall back to my old image?

Is that possible or makes it sense.

In short. Valetudo 0.3.0 has been installed and somehow fell back to 0.2.3.

Hypfer, 2019-04-07:
The robot does revert its firmware if something doesn't work so I guess that the new firmware image had some issues.

Did you follow the new guide? If you're using an old dustcloud checkout make sure to `git pull` to use the latest version of the imagebuilder

thundergreen, 2019-04-07:
@Hypfer  I deleted all directories and created everything from scratch. So there sholdn't be the problem.. but i cannot download th image from the docs.. i am actually via vpn via sweden online .. should i use another country?

thundergreen, 2019-04-07:
Unfortunately no change .. see here:

[image]

new firmware version flashed 3468 but no controls possible

--> rebooted: 

[image]

thundergreen, 2019-04-08:
Finally it set it self back to factory settings now..starting from scratch

cryptomilk, 2019-04-08:
The robot has an A/B system. If something goes wrong with the new flashed system it will boot to the old one which was working. Did you update the imagebuilder.sh script? You need the latest version ...

thundergreen, 2019-04-08:
I comletely started from scratch for updating to latest valetudo.. Two times it failed.. where could i see debig logs or something helpful?

before it reset to previous woring valetudo version now to factory reset

thundergreen, 2019-04-08:
EDIT: Now I am confused: I connected the robo with the app somehow to the wifi. And I saw that valetudo 0.3.0 is installed and running well. BUT i have the chinese language. Is it possible to change the language without flashing the device again? I took the language file english but somehow it doesn't take it into account. Anyway. The vacuum is still rooted and valetudo is running well.

CrazYoshi, 2019-04-08:
> EDIT: Now I am confused: I connected the robo with the app somehow to the wifi. And I saw that valetudo 0.3.0 is installed and running well. BUT i have the chinese language. Is it possible to change [...]

Same issue here. Flashed twice but robot speak chinese every time

fvollmer, 2019-04-08:
It might be related to the image builder: https://github.com/dgiese/dustcloud/issues/208

cryptomilk, 2019-04-09:
See the FAQ how to install a differnt language https://github.com/Hypfer/Valetudo/wiki/FAQ

This will also survive a new flash ...

closed 2019-04-12 by thundergreen

Hypfer, 2019-04-18:
I've also experienced a rollback of the firmware to 0.2.3

But.. why?