Hypfer/Valetudo #160: mqtt custom command

reaper7, 2019-04-08:
As I understand, current mqtt implementation does not support yet **custom command** option . I think there's only a declared topic?
Is planned a full implementation (for calling, for example, zone cleaning)?

Hypfer, 2019-04-08:
Problem is that the homeassistant mqtt vacuum doesn't support parameters yet so it's pretty useless to support custom commands.
See:
https://github.com/home-assistant/home-assistant/blob/dev/homeassistant/components/mqtt/vacuum.py#L511

If someone could fix that, I'll start fully implementing it in valetudo.

reaper7, 2019-04-08:
Yes, I know about it:
https://community.home-assistant.io/t/pass-send-command-params-to-mqtt-vacuum/109913
but there is the possibility of publishing custom mqtt msg from hass scripts or automations

Uzairhass, 2019-04-08:
Hi

reaper7, 2019-04-09:
It seems to me that the implementation should look like this:
https://gist.github.com/reaper7/630305cc972eec2090217f12b19afcfc

added:
line 25 - supported METHODS const
line 132 - subscribe custom_command topic
line 298 - parse json
line 345 - execute goto
line 350 - execute zone cleaning

Unfortunately, I'm not familiar with coding in javascript, it's just a concept.

assuming we send custom commands via mqtt in json:
```
{"method":"app_goto_target", "params":[24200,20200]}

{"method":"app_zoned_clean", "params":[[26234,26042,27284,26642,1]]}
```

reaper7, 2019-04-12:
new version which supports typical xiaomi commands like:
```
{"method":"app_goto_target", "params":[24200,20200]}
{"method":"app_zoned_clean", "params":[[26234,26042,27284,26642,1]]}
```
as well as simple mqtt commands:
```
{"app_goto_target":[24200,20200]}
{"app_zoned_clean":[[26234,26042,27284,26642,1]]}
```
version without logging:
https://gist.github.com/reaper7/18c157987a08c0dc67762e171583d571

and with logging to to /var/log/upstart/valetudo.log:
https://gist.github.com/reaper7/65f9bb0daba346bc1c2d3c07e445cf86

logs looks like:
```
Robot connected
MQTT: custom msg received: {"method":"app_goto_target","params":[27252,26773]}
MQTT: custom msg parsing OK
MQTT: custom msg xiaomi style
MQTT: custom msg method valid
MQTT: custom msg Command:app_goto_target
MQTT: custom msg Params :27252,26773
MQTT: custom msg received: {"method":"app_zoned_clean","params":[22769,22266,27252,26773]}
MQTT: custom msg parsing OK
MQTT: custom msg xiaomi style
MQTT: custom msg method valid
MQTT: custom msg Command:app_zoned_clean
MQTT: custom msg Params :22769,22266,27252,26773
```

Hypfer, 2019-04-13:
Good job 👍

However I'd rather use non-xiaomi-specific commands since we can do some transformations inbetween and this whole thing could at some point become generic.

There could be a `raw-command` command for users who explicitly want to run miio commands.
python-miio does this: https://python-miio.readthedocs.io/en/latest/vacuum.html

xoxys, 2019-04-30:
@Hypfer Custom commands should be possible now with home assistant or am I wrong?
https://www.home-assistant.io/components/vacuum.mqtt/#send-custom-command
https://github.com/home-assistant/home-assistant/blob/dev/homeassistant/components/vacuum/__init__.py#L53-L55

Hypfer, 2019-04-30:
@xoxys Well it's not released yet and will change. It does work, though.

I should really start working on feature branches

xoxys, 2019-04-30:
It is not released in HA? Or Valetudo?

xoxys, 2019-04-30:
What do you think about starting to use milestones or any other form of a roadmap. I am not a bin javascript/nodejs programmer (more apython guy) but maybe I can do some small things at first.

Hypfer, 2019-04-30:
Both HA as well as Valetudo

xoxys, 2019-04-30:
Hm, I see the requirements also in 0.92.1 release Tag
https://github.com/home-assistant/home-assistant/blob/0.92.1/homeassistant/components/vacuum/__init__.py#L53-L55

What am I  missing?

Hypfer, 2019-04-30:
https://github.com/home-assistant/home-assistant/blob/0.92.1/homeassistant/components/mqtt/vacuum.py#L511

It is indeed supported in HA 0.92.1

hagenuck1, 2019-04-30:
Does anybody know if the send command Integration in Home assistant is supporting templates? 

With the xiaomi vacuum Integration send command was possible, but it doesn’t support template values.

AS137430, 2019-05-02:
#204 It's working in HA 0.92.1 for me with a code change. https://github.com/Hypfer/Valetudo/blob/master/lib/MqttClient.js#L336
```
 this.vacuum.startCleaningZone(zone[1], () => {});
```
For the zones variant of the send_command, it should be an array of the same format as app_zoned_clean's parameter:
```
[[26234,26042,27284,26642,1]]
```

closed 2019-06-11 by Hypfer

darkkatarsis, 2019-11-07:
Is it possible to send a vacuum cleaner under specific corridates rather than an ID zone / spot? If not, will this be added in the future?