Hypfer/Valetudo #161: no Mapupdate by using Reverseproxy

gary822, 2019-04-08:
when you are using an Apache2 ProxyPass with HTTPS there are no live map update.
I'm using the following config:

<VirtualHost *:80>
  ServerAdmin webmaster@localhost
  DocumentRoot /var/www/

  # HTTP zu HTPPS umschreiben
  RewriteEngine On
  RewriteCond %{HTTPS} off
  RewriteRule (.*) https://%{HTTP_HOST}%{REQUEST_URI} [R=301,L]
</VirtualHost>

  #staubi
  ProxyPass "/staubi" "http://192.168.1.6"
  ProxyPassReverse "/staubi" "http://192.168.1.6"

Hypfer, 2019-04-08:
Yep. It's a known issue fixed in the current master but not in 0.3.0

Should work in the next release
See https://github.com/Hypfer/Valetudo/commit/d8c38176120333f2fcc8ee193f83670ebf237b89

gary822, 2019-04-08:
Ah thanks for the feedback.
will it be in a beta version?

Hypfer, 2019-04-08:
It will be in the next release

closed 2019-04-25 by Hypfer

Snobs76, 2019-06-13:
My working configuration with live map update. 
Server version: Apache/2.4.25 (Debian)
Following mods must be enabled:

a2enmod proxy
a2enmod proxy_http
a2enmod proxy_wstunnel
`<VirtualHost *:443>
  ServerName vacuum.meinedyndnsadresse.de

  SSLEngine on
  SSLCertificateFile /etc/letsencrypt/live/vacuum.meinedyndnsadresse.de/fullchain.pem
  SSLCertificateKeyFile /etc/letsencrypt/live/vacuum.meinedyndnsadresse.de/privkey.pem
  Include /etc/letsencrypt/options-ssl-apache.conf

  RequestHeader    set X-SCHEME http

  RewriteEngine On
  RewriteCond %{REQUEST_URI}  ^/    [NC,OR]
  RewriteCond %{QUERY_STRING} transport=websocket    [NC]
  RewriteCond %{HTTP:UPGRADE} =websocket [NC]
  RewriteRule /(.*) ws://192.168.178.100/$1 [P,L]

  ProxyPass / http://192.168.178.100/
  ProxyPassReverse / http://192.168.178.100/

<Location "/">
  AuthType Basic
  AuthUserFile /etc/apache2/.passwd
  AuthName "Please login."
  order deny,allow
  allow from all
  require valid-user
</Location>

</VirtualHost>`