Hypfer/Valetudo #164: Recommended  Firmware & change of IP

rassaei, 2019-04-09:
Is there any documentation of why which firmware version is recommended and which ones are not? I'm asking as dustcloud wiki and valetudo wiki recommend different FWs for the s50.

Also: is the IP of the vacuum set in the images? Why is it in another subnet? (192.168.178.0 vs 192.168.8.0) Is it not set via DHCP?

Hypfer, 2019-04-09:
192.168.8.1 is the IP of the robot when it's in AP mode. If it's connected to your wifi it picks up an address via DHCP.

Regarding firmwares: I have absolutely no idea. I'm using 1748 on my Gen2 if that helps you.

rassaei, 2019-04-09:
Thx for the IP info, I was very confused. A lot changed in the last couple of weeks :D

For the firmwares: Also very confusing. I tried to make some sense by checking the wiki diffs but the numbers just change (sometimes they get higher, sometimes lower) without any explanation :D Thx anyway. Maybe someone else can help. @cryptomilk maybe? He did some of the changes.

onovy, 2019-04-27:
I'm using 1768 and it is just newer bugfix release after 1748. Works fine with Valetudo.
[image]
[image]

Hypfer, 2019-04-28:
@onovy Can we access those changelogs somehow without using the app?

closed 2019-04-28 by Hypfer

onovy, 2019-04-28:
@Hypfer I don't think so. These screenshots are from Facebook roborock group.

Hypfer, 2019-04-28:
Can you share them somewhere or just add the changelogs to the dustcloud wiki?