Hypfer/Valetudo #165: Voicepacks

Hypfer, 2019-04-09:
Since there seems to be some demand for custom voicepacks, I'd like to write down my thoughts on how Valetudo should approach this topic.

It might make sense to split this into multiple phases:

**Phase 1**
A simple file upload dialog where the user can pick a local language pack file and upload it through valetudo to the robot.

**Phase 2**
In the long run, I'd love to see a seperated github repo (maybe *valetudo-voice*) which contains curated and free voicepacks. Valetudo should fetch the latest list of all available voicepacks in said repo and render a nice view where the user can simply tap on one to install it.
It might make sense to also add preview audio files and make them playable in said view.

There should be a template voicepack for *valetudo-voice* so that people who want to contribute a new pack know which files they need and what they do. Then do some Travis CI stuff on Pull Requests to build the voicepack archives in the robot format.

thundergreen, 2019-04-09:
I totally agree. We might add other languages than English and German. Like French Italien etc.

Hypfer, 2019-04-09:
Since the language of the voicepack is controlled by `/mnt/default/roborock.conf` I wonder what happens if you change `language=prc` to `language=someRandomString`.

If that just loads the files from someRandomString, we might have a way to manage multiple packs for a single language or packs for no real languages at all.

TheRealMoeder, 2019-10-08:
Hi! I already realized an automated voicepack generator for a different project utilizing pythonanywhere, github and travis. As it was only little work, I quickly ported this to be used with valetudo. The voices used are mostly high quality versions from MacOS X, so the result is pretty good with many languages available. You can try the service [here](http://moeder.pythonanywhere.com/rockrobo), the resulting pkgs are pushed back to my [github.io page](http://therealmoeder.github.io/rockrobo.html). It usually takes only 2-3 minutes for the voicepack to get finished.

TheRealMoeder, 2019-10-08:
@Hypfer thanks for the thumbs up ;) Do you think the voice pack generator is something which should be more refined for valetudo? I could try to spare some time. The method of python triggering Travis CI is pretty usable, but I still dislike the pushback to GitHub.io, as it is more of a hack. I guess I could also store the files on pythonanywhere and give a hashed link for download, but filestorage and bandwidth are very limited on the free plan. I haven't found any other suitable free service. Any recommendations?

OzGav, 2020-10-03:
Hi,
Just installed this on my roborock v1 which is awesome. I just want to ask though about changing to English voice. The docs say to edit the file `/mnt/default/roborock.conf` but I don't seem to have this file on my vacuum. Is it somewhere else now or am I missing something?

raid1, 2020-10-15:
I recorded and made a bavarian (German dialect) package.
Where could I upload this for others who might enjoy this?
TheRealMoeder's service seems to be down....

thundergreen, 2020-10-16:
on my server! :) I have a simple fileserver feel free tu upload there :) https://droppy.ddns.me/

OzGav, 2020-10-16:
So are you guys saying you have access to the `roborock.conf `file? Which robot are you using?

jneuhauser, 2020-10-17:
@raid1 I'd like to have your bavarian voice pack... Why not sharing it over google drive or the like?

raid1, 2020-10-17:
> @raid1 I'd like to have your bavarian voice pack... Why not sharing it over google drive or the like?

Sure, here you go: https://cloud.spiegl.de/index.php/s/AQ6dJqWXCn8tkY5
You are free to give it to others if you like (public domain)
I did not translate all phrases but only the 36 most used ones. Errors etc are still in English.

Have fun with it!

raid1, 2020-10-17:
> on my server! :) I have a simple fileserver feel free tu upload there :) https://droppy.ddns.me/
done. You are free to give it to others if you like (public domain)
I did not translate all phrases but only the 36 most used ones. Errors etc are still in English.

> I uploaded the German one.
I only see mine

brujoand, 2020-11-16:
The [generate_audio.py](https://github.com/dgiese/dustcloud/tree/master/devices/xiaomi.vacuum/audio_generator) script from dustcloud worked great for me. I just had to change the voice name and bom! :)

OzGav, 2020-11-17:
Still wanting to know how to change the voice if I can't find roborock.conf

brujoand, 2020-11-18:
I haven’t looked for that file but settings -> Sound & voice -> Upload works fine for a simple swap.

OzGav, 2020-11-19:
I believe all the voices are on there because I can see a pile of WAB files under `/mnt/updbuf/opt/rockrobo/resources/sounds/en/` but I just can't work out how to activate them. If I wanted to upload an english pack again as per your suggestion is there one somewhere?

closed 2021-01-09 by Hypfer

tasmota-benutzer, 2021-03-14:
> Sure, here you go: https://cloud.spiegl.de/index.php/s/AQ6dJqWXCn8tkY5
> You are free to give it to others if you like (public domain)
> I did not translate all phrases but only the 36 most used ones. [...]

Servus, I tried to install the pkg via valetudo web interface, but it failed.
The regular germen voice pack worked out of the box. Any ideas?

raid1, 2021-03-15:
> Any ideas?
Uhm, actually no. No one else has reported this so far.

tasmota-benutzer, 2021-03-15:
Can I somehow extract the wav files out of that pkg?
It seems not to be a tar.gz or something like that...

StephanHoetzl, 2021-04-01:
> Sure, here you go: https://cloud.spiegl.de/index.php/s/AQ6dJqWXCn8tkY5
> You are free to give it to others if you like (public domain)
> I did not translate all phrases but only the 36 most used ones. [...]

Hi raid1, awesome job with the bavarian voice pack. Do you have a list of all commands? In the lists I can find the commands for e.g. room cleaning seem to be missing.

raid1, 2021-04-02:
Glad you like it. My kids are not so fond of their dad's voice in the cleaner.  :-)

Which command/file are you missing? I thought I had "translated" all important files.
Only some errors are still in English on mine...

StephanHoetzl, 2021-04-09:
I probably did not express myself quite thoroughly/clearly. 

Your voicepack ist not missing any commands I would like to have. However, I tried to create an own soundpack and succeeded in principle. But I do not know how the individual *.wav-files have to be named. The lists I could find show "bin_in.wav; charging.wav; start.wav; stop.wav;...etc." but no commands for e.g. the sound "starting room cleaning".
You seem to have a complete list of commands or maybe I am just missing something :)

tasmota-benutzer, 2021-04-09:
I downloaded the complete folder from the vacuum cleaner in my language (german) and listened to each file. After that, I recorded my own voice with my "special slang" and replaced the wav file. Maybe that way it works out for you?

raid1, 2021-04-09:
I did it exactly as @tasmota-benutzer described. Worked like a charm although I would also have preferred to have a complete list as a reference.