Hypfer/Valetudo #166: Problems after upgrading to 0.3.0

dh-harald, 2019-04-09:
Hi,

I'm using Valetudo for a while without issues, but unfortunately after upgrading to 0.3.0 I've got several issues.
Currently my issues:
- no map... I've got just the "robot" icon.
- couple of status are missing from mqtt in HA: battery, status text
Here's my all valetudo related datas in mqtt:
```
valetudo/xiaomi_vacuum_cleaner/attributes {"cleanTime":"50.8","cleanArea":"2589.4","cleanCount":164,"mainBrush":"249.0","sideBrush":"149.0","filter":"99.0","sensor":"14.7"}
homeassistant/vacuum/valetudo_xiaomi_vacuum_cleaner/config {"name":"xiaomi_vacuum_cleaner","supported_features":["turn_on","pause","stop","return_home","battery","status","locate","clean_spot","fan_speed","send_command"],"command_topic":"valetudo/xiaomi_vacuum_cleaner/command","battery_level_topic":"valetudo/xiaomi_vacuum_cleaner/state","battery_level_template":"{{ value_json.battery_level }}","charging_topic":"valetudo/xiaomi_vacuum_cleaner/state","charging_template":"{{value_json.charging}}","cleaning_topic":"valetudo/xiaomi_vacuum_cleaner/state","cleaning_template":"{{value_json.cleaning}}","docked_topic":"valetudo/xiaomi_vacuum_cleaner/state","docked_template":"{{value_json.docked}}","error_topic":"valetudo/xiaomi_vacuum_cleaner/state","error_template":"{{value_json.error}}","fan_speed_topic":"valetudo/xiaomi_vacuum_cleaner/state","fan_speed_template":"{{ value_json.fan_speed }}","set_fan_speed_topic":"valetudo/xiaomi_vacuum_cleaner/set_fan_speed","fan_speed_list":["min","medium","high","max"],"send_command_topic":"valetudo/xiaomi_vacuum_cleaner/custom_command","json_attributes_topic":"valetudo/xiaomi_vacuum_cleaner/attributes"}
homeassistant/camera/valetudo_xiaomi_vacuum_cleaner_map/config {"name":"xiaomi_vacuum_cleaner_map","unique_id":"xiaomi_vacuum_cleaner_map","topic":"valetudo/xiaomi_vacuum_cleaner/map"}
```
I've started with firmware 1720, I just replaced the binary and I've restarted the service... Than I tried to upgrade the firmware to 1748 (used the guide in dustcloud), but the problems are still exists...
Any ideas?

Hypfer, 2019-04-09:
The robot is not connecting to the cloud interface of valetudo. Something about the traffic redirection in /etc/hosts and /etc/rc.local

dh-harald, 2019-04-09:
On the web page, it looks good, the status is changing, for example cleaning, idle, returning to home, that part is working. Just not appear in mqtt
Originally, I've used the original /etc/hosts (not your upstream from the git repo), but after I start using yours, the only difference is, that the robot can't upload the map to the xiaomi cloud, so I can't check the map in the app either... But still no statuses/map in mqtt.

Hypfer, 2019-04-09:
Currently, the status page uses the local interface so thats why that works.

dh-harald, 2019-04-09:
/etc/rc.local & /etc/hosts are exactly the same like from your upstream repository...
rc.local runs fine, because I could see the iptables rules

Hypfer, 2019-04-09:
Try rebooting if you haven't done so already. If you reroute on a running system the new rules do not apply.

I think it's network related.

dh-harald, 2019-04-09:
I've rebooted my device, and now it looks working... I don't know, what's the difference... Thanks for help

closed 2019-04-09 by dh-harald

Hypfer, 2019-04-09:
I assume it's because the connection has already been established or something like that.

xoxys, 2019-04-22:
@Hypfer can we add a notice to the wiki that the map only works AFTER you added the device to a wifi? I needed many hours to figure out that (for an unknown reasons) the map will not work while I am connected through the default network.

Hypfer, 2019-04-23:
@xoxys I didn't know that it does but that makes sense since we're now using the cloud interface and that obviously requires the robot to be provisioned (wifi configured).

Will you add it to the wiki?

xoxys, 2019-04-23:
@hypfer sure :) thanks for clarification

Hypfer, 2019-04-23:
@xoxys So.. the webinterface itself did work in AP mode for 0.3.1?

There apparently were some issues with that:
https://github.com/Hypfer/Valetudo/issues/181#issuecomment-485423221

xoxys, 2019-04-23:
@Hypfer Did not test it with 0.3.1 but with 0.3.0 it works. I am using firmware 3.3.9_001810 for gen2.