Hypfer/Valetudo #167: Impossible to change Valetudo region

CrazYoshi, 2019-04-09:
Hi, 
I flashed my rockrobo yesterday with the latest version of Valetudo from github but I have an issue that was not present in the previous version. Even if I have applied the language pack for english I still have Language and Location set to prc, Timezone to Asia/Shanghai and Logserver to awsbj0.fds.api.xiaomi.com
How can I solve this issue?

cryptomilk, 2019-04-09:
See FAQ

CrazYoshi, 2019-04-09:
I don't have any file /mnt/default/roborock.conf on my rockrobo

cryptomilk, 2019-04-09:
So you have a Gen1 robot?

CrazYoshi, 2019-04-09:
yes

cryptomilk, 2019-04-09:
That's a detail which should be in the initial description. I dunno how this is configured in Gen1 I guess you have to find it on your own. grep is your friend :-)

Franzl123, 2019-04-09:
see https://github.com/dgiese/dustcloud/wiki/Vacuum-Robots-CCC-to-CE-conversion

fvollmer, 2019-04-09:
I just successfully changed the language by creating the
`/mnt/default/roborock.conf`
file. Not only `en` works, but I also could use `de`! (The robot uses the
soundfiles from `/opt/rockrobo/resources/sounds/de`)

cryptomilk, 2019-04-09:
If you install a sound pack then it is put into /mnt/data and uses that one.

fvollmer, 2019-04-09:
Wouldn't it me useful if the image builder would put the sound files default at this place? I would imagine that this would avoid problems due to different robot regions/languages?

cryptomilk, 2019-04-10:
`/mnt/data` is an extra partion, it isn't included in the FW.

cryptomilk, 2019-04-10:
The more you put into the FW the less space we have for e.g. Valetudo.

Franzl123, 2019-04-10:
hmm,
an idea would be
- to add an new switch in imagebuilder.sh (like "-language | -l"), where the "-soundfile" is stored (eg. /opt/rockrobo/resources/sounds/de) and then delete all others (en, tw, prc).

- when "-language" is not set, the "-soundfile" goes to /opt/rockrobo/resources/sounds/en, like now, but delete all others too.

- to activate the new language, the robot must write the given "-language" (or "language=en") to /mnt/default/roborock.conf (see the ccc-to-ce thing) on first boot (via /etc/rc.local)

ps: i don't know, if the deleting of the unused soundfiles is ok for the robot!

cryptomilk, 2019-04-10:
We should just support uploading sound packs as suggest in #165. Once a soundpack has been uploaded it will just always be used, even if you upgrade the firmware.

Franzl123, 2019-04-10:
yes, this will be the best/cleanest option of course.
it was just an idea, because imagebuilder don't do it in the moment :)

fvollmer, 2019-04-10:
Another idea for the imagebuilder:
just copy the files to `/opt/rockrobo/resources/sounds/all` and use symlinks for `/opt/rockrobo/resources/sounds/en` and `/opt/rockrobo/resources/sounds/prc`. This way the sound files should always work.

Nebukad93, 2019-04-10:
Image builder already allow you to upload a new language package. The only this is that it will be considered like the English language

Franzl123, 2019-04-10:
but the robot must set to english language via mii-app. my robot for example (and in the beginning of this issue the one of CrazYoshi) was not set to english (and had never an connection to the mii-app) and so it don't used the sound-files. now, i set it via `/mnt/default/roborock.conf` to english and it works.

but the symlinking-idea from fvollmer is a good one! no struggeling with config files. the only thing i would change is to put the sounds in `/opt/rockrobo/resources/sounds/en` directly and link to `/opt/rockrobo/resources/sounds/prc`, so there is no space wasted

CrazYoshi, 2019-04-12:
Another issue that i have at the moment is that the map doesn't update during a cleanup.
It remains like this [https://imgur.com/C4qLCsK](url)

cryptomilk, 2019-04-12:
This bug is about region changes and not about the map. Please go to #172 !

CrazYoshi, 2019-04-12:
> This bug is about region changes and not about the map. Please go to #172 !

ok, no problem.
About region change I follow this: [https://github.com/dgiese/dustcloud/wiki/Vacuum-Robots-CCC-to-CE-conversion](url) but when I go to settings now it never stop loading and the page doesn't show me name, bom, location, timezone, etc. Why?

Franzl123, 2019-04-12:
A reboot should help

closed 2019-06-04 by Hypfer