Hypfer/Valetudo #168: Adding zone coordinates function

mar565, 2019-04-09:
Hello folks,
I would like to acces saved zones from a home automation system (expl. ioBroker).
I've add a few zones using the interface and the 'rockrobo/zone' site.
Now I'd like to acces the coordinates to feed the parametes of the 'mihome-vaccum adapter' and ultimatly make it run via vis.
The idea is to have a function that revcievs the name of a saved zone and return an array with the coordinates (since there are zones made out of multiple rectangle sections)

cryptomilk, 2019-04-10:
Zone support is still work in progress, see #157

vene4ka, 2019-04-14:
GET request to http://rockrobo/api/get_config returns object with saved zones in _areas_ key among other.
`curl 'http://rockrobo/api/get_config' -H 'DNT: 1' -H 'Accept-Encoding: gzip, deflate' -H 'Accept-Language: uk-UA,uk;q=0.9,ru;q*/*' -H 'Referer: http://rockrobo/' --compressed
`

Hypfer, 2019-04-14:
@vene4ka not for long. Please don't rely on that.

closed 2019-04-28 by Hypfer