Hypfer/Valetudo #171: oddly rotated map

Technikte, 2019-04-10:
Hi,
like you see in the first screenshot the map is somewhat oddly slightly rotated after I done 
`echo -n "1" > /mnt/data/rockrobo/lab.cfg` and manually turn off/on the robot to reboot it

Firmeware: 3.3.9
Firmware build: 001768
Valetudo: 0.3.0

[image]

[image]

Is there anything that I can do to reset the map or just to correct the angle ?

cryptomilk, 2019-04-10:
It should fix the rotation on the next run, if not you can reset the map with:

`mirobo raw-command reset_map`

Technikte, 2019-04-10:
> 
> 
> It should fix the rotation on the next run, if not you can reset the map with:
> 
> `mirobo raw-command reset_map`

Wow, that was a quick answer..can I trigger the map reset somehow directly on the robot itself via ssh ?

cryptomilk, 2019-04-10:
I'm sure you can if you install python3-miio :-)

Technikte, 2019-04-10:
It seems like a rerun has purged the map and now a map at correct angle got generated.
Fixed, I will close the issue.

closed 2019-04-10 by Technikte