Hypfer/Valetudo #172: Problem with mapfunction

marooram, 2019-04-11:
I followed your wiki and installed valetudo onto the roborock (2nd gen with your recommended FW 1748). Currently I never get a map (all other functions work well), also tested "cleaning" and "restart robot" as said in the wiki. I only see the big robot.. what could be the problem and how could I solve it?

reaper7, 2019-04-12:
I noticed something like that yesterday
but after _sudo reboot_ the map came back

CrazYoshi, 2019-04-12:
> I followed your wiki and installed valetudo onto the roborock (2nd gen with your recommended FW 1748). Currently I never get a map (all other functions work well), also tested "cleaning" and [...]

I have the same issue with my gen1 rockrobo. In the previous version of Valetudo it works correctly but after the update to 3.0 map is always empty.

tmrealive, 2019-04-12:
I have the same problem when upgrading to 0.3.0, gen1

hyperbit00, 2019-04-12:
same problem here, works like a charm with V0.2.0, with 0.3.0 only a big robot

Try'd it with compiling v0.3.0 by my self and with installing rooted firmware with integrated Valetudo (wiki instructions)

vevsvevs, 2019-04-12:
Guys, check your `/etc/hosts` and `/etc/rc.local` files, they must contain strings from this [deployment examples](https://github.com/Hypfer/Valetudo/tree/master/deployment/etc). Also remove `mapSettings` entries from `valetudo.conf`.

hyperbit00, 2019-04-12:
> Guys, check your `/etc/hosts` and `/etc/rc.local` files, they must contain strings from this [deployment examples](https://github.com/Hypfer/Valetudo/tree/master/deployment/etc). Also remove `mapSettings` entries from `valetudo.conf`. And don't forget to reboot your vacuum after all ;)

that was the solution

marooram, 2019-04-12:
> > Guys, check your `/etc/hosts` and `/etc/rc.local` files, they must contain strings from this [deployment examples](https://github.com/Hypfer/Valetudo/tree/master/deployment/etc). Also remove `mapSettings` entries from `valetudo.conf`. And don't forget to reboot your vacuum after all ;)
> 
> that was the solution
> any solution to integrate it to IOBROKER? (maybe iframe?)

Mayybat following could help you:
https://github.com/MeisterTR/ioB-scripts.git

marooram, 2019-04-12:
> Guys, check your `/etc/hosts` and `/etc/rc.local` files, they must contain strings from these [deployment examples](https://github.com/Hypfer/Valetudo/tree/master/deployment/etc). Also remove `mapSettings` entries from `valetudo.conf`. And don't forget to reboot your vacuum after all ;)

Worked for me too, thank you!

closed 2019-04-12 by marooram

roflcoopter, 2019-04-13:
> Worked for me too, thank you!

Where is the valetudo.conf located? Or do you mean config.json?

reaper7, 2019-04-13:
/etc/init

vevsvevs, 2019-04-13:
> Where is the valetudo.conf located? Or do you mean config.json?
Oh, of course i meant `/mnt/data/valetudo/config.json`, sorry :) Post corrected.