Hypfer/Valetudo #173: Zone cleanup with /zone api

pszafer, 2019-04-12:
Hi,
So I am trying new subpage http://IP/zone and got several problems with that.
Vacuum hangs if room is bigger (20m2), valetudo webpage is not responding.
I don't see anything in valetudo logs, but I saw that gzip and player processes tooked 100% CPU when I logged in via ssh.
After a while on both maps (map in webui and /zone) valetudo thinks that vacuum is nearby charging station, while it's far away... Clicking on vacuum button go back to dock, vacuum is going back.

Valetudo ver 0.3.0
Vacuum gen 2.

pszafer, 2019-04-12:
Ok, think it is duplicate of #174 
have same error eventually:
```
buffer.js:38^M
    super(arg1, arg2, arg3);^M
    ^^M
^M
RangeError: Array buffer allocation failed^M
    at new ArrayBuffer (<anonymous>)^M
    at typedArrayConstructByLength (<anonymous>)^M
    at new Uint8Array (native)^M
    at new FastBuffer (buffer.js:38:5)^M
    at Function.Buffer.alloc (buffer.js:245:10)^M
    at new Buffer (buffer.js:156:19)^M
    at new exports.PNG (/snapshot/Valetudo/node_modules/pngjs/lib/png.js:20:5)^M
    at Jimp.getBuffer (/snapshot/Valetudo/node_modules/jimp/dist/main/utils/image-bitmap.js:237:19)^M
    at Jimp.<anonymous> (/snapshot/Valetudo/lib/Tools.js:142:39)^M
    at Timeout._onTimeout (/snapshot/Valetudo/node_modules/jimp/dist/main/index.js:258:25)^M
```

Would be possible to add timestamp to logs?

Hypfer, 2019-04-12:
#154

pszafer, 2019-04-12:
what about player process? It is taking a 40-50% of memory and around 15% of CPU.

aprosvetova, 2019-04-14:
@pszafer that's normal. player process is the main roborock process that takes care of cleaning process and dealing with hardware elements like sensors, motors, the lidar, etc. It has nothing to do with Valetudo.

closed 2019-06-04 by Hypfer