Hypfer/Valetudo #174: Valetudo crashing

blakeblackshear, 2019-04-12:
Valetudo frequently crashes for me with the following error message:

```
Webserver running on port 80
Robot connected
buffer.js:38
    super(arg1, arg2, arg3);
    ^

RangeError: Array buffer allocation failed
    at new ArrayBuffer (<anonymous>)
    at typedArrayConstructByLength (<anonymous>)
    at new Uint8Array (native)
    at new FastBuffer (buffer.js:38:5)
    at Function.Buffer.alloc (buffer.js:245:10)
    at new Buffer (buffer.js:156:19)
    at new exports.PNG (/snapshot/Valetudo/node_modules/pngjs/lib/png.js:20:5)
    at Jimp.getBuffer (/snapshot/Valetudo/node_modules/jimp/dist/main/utils/image-bitmap.js:237:19)
    at Jimp.<anonymous> (/snapshot/Valetudo/lib/Tools.js:142:39)
    at Timeout._onTimeout (/snapshot/Valetudo/node_modules/jimp/dist/main/index.js:258:25)
Loading configuration file: /mnt/data/valetudo/config.json
Dummycloud is spoofing 203.0.113.1:8053 on 127.0.0.1:8053
Webserver running on port 80
```

I am guessing this happens because my floorplan is quite large.

Hypfer, 2019-04-12:
Looks like we might need to drop PNG generation altogether

blakeblackshear, 2019-04-12:
I think it is surpassing what is reasonable to do directly on the vacuum itself. When I watch top during runs, it seems like player and valetudo are fighting each other for resources. I was thinking about trying to find a way to offload as much as possible to a separate server.

Hypfer, 2019-04-12:
Yep. The first map implementation was client-side canvas only due to exactly that.
Then, people wanted to integrate their map into various home automation software so thats why this server-side rendering even exists at all. :/

Good news is that at least Home Assistant should be rather easy to support since we can just put the drawing logic into a custom lovelace card. As for FHEM it could be possible to take the png drawing code and run it somewhere else with the map data from MQTT.

There aren't resource issues though since player runs at -20 nice and valetudo is limited to around 80mb of ram and has an OOM score of 1000 so player always wins this.

blakeblackshear, 2019-04-12:
It looks like most of the communication is happening with through the miio interface anyway right? What requires the process to run on the actual vacuum at this point?

Hypfer, 2019-04-12:
Valetudo reads the local token, cloudkey and deviceID from fs. Also there might be new features which also require direct hardware access.

Of course it's possible to run Valetudo somewhere else. I'm doing exactly that while developing. It's just neat to have a self-contained vacuum which provides well-defined interfaces to interact with it locally.

blakeblackshear, 2019-04-12:
Great. I will look into running it on a separate server. Would be nice to have support for multiple vacuums at one url too.

Hypfer, 2019-04-12:
I actually thought about this for the e3 vacuums. In theory it should be possible to run them with valetudo if you manage to aquire the cloudkey and redirect their traffic.

Still, I think that running directly on the vacuum should be the main focus of valetudo. It's also much easier to just move the PNG drawing somewhere else.

blakeblackshear, 2019-04-12:
Agreed.

Hypfer, 2019-04-12:
I've moved the PNG Generation into a seperate project:
https://github.com/Hypfer/ICantBelieveItsNotValetudo

You will need the latest valetudo master for the parsed Map data as JSON via MQTT.

K-os, 2019-04-18:
I also have the problem of Valetudo being killed by the OOM-killer, but I don’t understand why it also halts the cleaning process. Is it because the player can’t reach the “cloud” any more?
This makes the version 0.3 unusable for me.
Would it be possible to reimplement the client side rendering mentioned above? I don’t need the map to be available via MQTT, but I like the live map in the web interface.

Hypfer, 2019-04-18:
@K-os I guess it would make sense to release 0.3.1 shortly where MQTT PNG Map rendering is removed from Valetudo and moved to the sideproject

Hypfer, 2019-04-21:
https://github.com/Hypfer/Valetudo/releases/tag/0.3.1

closed 2019-04-21 by Hypfer