Hypfer/Valetudo #177: Gen 3 (T6) Support

Nixon506E, 2019-04-15:
Opening this issue to start work on rooting and supporting the new functionality in the T6 model.

dugite-code, 2019-04-16:
You might be better asking in the [dustcloud](https://github.com/dgiese/dustcloud) repo about root for the T6. That said from the marketing material it has an allwinner cpu so it might be very similar to the two existing models.

The main reason we have root on the Gen1 and 2 is that the firmware has a static password: “rockrobo”. It will be interesting to see if they change this as they haven't appeared to be openly hostile to the OS community

closed 2019-04-25 by Hypfer

MickL, 2019-06-15:
T6 is available in the EU now know as the S6. They started officially in Germany, France and Spain as of now. You might rename this issue to `T6/S6`. Also @Hypfer please reopen this issue and label it as postponed until https://github.com/dgiese/dustcloud/issues/213 is solved

New features that need to be implemented in Valetudo, after the device is rooted, are:

- Assign and edit rooms on the map and display them with different colors
- Assign and edit districted zones
- Clean specific room

Hypfer, 2019-06-15:
`Also @Hypfer please reopen this issue and label it as postponed until dgiese/dustcloud#213 is solved`
You're not the one giving orders here. With this attitude you will be banned in no time.

Also don't mention me unless it is really important. I've unwatched this repo to not receive this email spam

MickL, 2019-06-15:
I dont give orders i thought this is an open source project. You closed this issue you should reopen. If you are the king of your empire ok but no one will contribute. And sorry for mentioning you princess @Hypfer

Hypfer, 2019-06-15:
lol