Hypfer/Valetudo #179: Unable to connect after upgrade to 0.3

micgus, 2019-04-18:
Heya folks. 
When building and flashing my gen2 with 0.3, I can no longer connect to the vacuum, downgrading again to any other earlier Valetudo works just fine.

Tried to base it off v11_001810 as well as v11_001748 but it's the same behaviour.
Any input? Any ideas on how to start troubleshooting?
Cheers.

ralf-ms, 2019-04-19:
Same for me, unable to connect, (even after rebooting: gen 1 robot, fw 3514, process valetudo is running). 
0.3.0 is unusable for me :-(, downgraded to 0.2.3 and again everything is working fine.

micgus, 2019-04-20:
> Heya folks.
> When building and flashing my gen2 with 0.3, I can no longer connect to the vacuum, downgrading again to any other earlier Valetudo works just fine.
>
> Tried to base it off v11_001810 as [...]

Resolved the issue.

Scraped the ubuntu VM and started over and now the build process actually builds a FW. 
For some reason OEM FW was flashed in my previous attempts.

ralf-ms, 2019-04-23:
Rebuilding and uploading firmware solved my problem too.
It's working now!

closed 2019-04-23 by Hypfer