Hypfer/Valetudo #181: Loaded firmware 3468.fullos successfully but can't connect to 192.168.8.1

bigjohns97, 2019-04-22:
Went through the install guide and built the firmware based off the image in the title and everything loaded successfully to the robo but I can't connect.

Is there something I am supposed to do to be able to connect to the web interface once the fw is loaded successfully?

fvollmer, 2019-04-22:
The web interface should be available after a successful firmware update
(if you are in the same network).

There was a known issue that resets the robot firmware to the old firmware
(without valetudo). Have you tried the recently released 0.3.1 valetudo
version? It contains a possible fix for this issue.

Went through the install guide and built the firmware based off the image

bigjohns97, 2019-04-22:
It's possible, I built the original build (on my desktop) before realizing that i was going to have to connect to the open WiFi with a laptop with 0.3.0 but i distinctly remember seeing and using the new 0.3.1 and using it. I don't have a problem rebuilding just to be sure but it might be a couple of days before i can do so.

I did notice that the first time i tried to load it the load took twice as long and it crashed at 96% which seemed to be some sort of a socket error. However when i ran it again immediately afterwards it completed and showed success.

You don't think that first crash has anything to do with it do you?

ivan98, 2019-04-22:
My gen2 just came in today, so [disclaimer] I'm a newbie. Flashed 2-3 times, nothing on port 22 nor 80 at 192.168.8.1. On the 4th flash, I put in multiple SSH keys and finally I can login. First check was Valetudo, it wasn't running. I manually run it. Use Chrome to configure Wi-Fi for the vacuum to connect to my home Wi-Fi. Reboot.

Now, Valetudo starts up upon reboot.

Checking /etc/init/valetudo.conf, it is saying it will start upon state change to up for the first network interface that is not local loopback. Now, I'm not super confident of Wi-Fi on Linux, so the state change might not have been detected. It may have got better after I configure it to connect to a regular SSID.

Hypfer, 2019-04-22:
@ivan98 Interesting report. I will check that soon.

Waiting for any interface to come up before the start of valetudo is a proposed fix for random rollbacks and loss of wifi credentials. It would be bad if that breaks AP Mode.

bigjohns97, 2019-04-22:
> My gen2 just came in today, so [disclaimer] I'm a newbie. Flashed 2-3 times, nothing on port 22 nor 80 at 192.168.8.1. On the 4th flash, I put in multiple SSH keys and finally I can login. First [...]

I might try this later on when i get home, how many keys did you specify? 

Could you post your flash string?

ivan98, 2019-04-23:
```shell
sudo modprobe loop
if [ ! -r patcher.py ]; then
  wget https://raw.githubusercontent.com/JohnRev/rrlogd-patcher/master/patcher.py
fi
sudo ../dustcloud/devices/xiaomi.vacuum/firmwarebuilder/imagebuilder.sh \
     --firmware=../firmware/v11_001810.fullos.pkg \
     --public-key=$HOME/.ssh/id_ed25519.pub \
     --public-key=ed25519-ilmac.pub \
     --valetudo-path=../valetudo \
     --disable-firmware-updates \
     --ntpserver=192.168.1.254 \
     --timezone=Asia/Singapore \
     --rrlogd-patcher=patcher.py \
     --enable-ruby \
     --replace-adbd
```
My flash script is above. My kernel's loop is compiled as a module, so I need modprobe prior.

enable-ruby is not needed, as chpasswd fails with "Cannot determine your user name", so user ruby is enabled with an unknown password, hence not used at all.

I specified two SSH keys, 1 from my Linux server, another from my iMac. When roborock is in unprovisioned state, my iMac connects to it, so iMac's SSH key is needed.

I build the image on my Linux, transfer over to my iMac, connect iMac to roborock's unprovisioned SSID, run flasher.py.

Hypfer, 2019-04-23:
Can confirm valetudo not starting on an unprovisioned robot.

Should be fixed now. If you're building images please redownload the valetudo.conf upstart file

bigjohns97, 2019-04-23:
I finally got the image to load all the way without timing out or crashing the script. Once the robo sounded off that it was doing an update i let it finish and then tried the web interface. Didn't work so i was able to SSH into it and start valetudo, now i can get to the web interface but anytime i try to join it to my WiFi it just hangs like it is going to join but it never joins.

bigjohns97, 2019-04-23:
Finally was able to get it to join WiFi after a reboot and manually start valetudo again, right now I am going to stand where I am at which is just being able to push the button to start cleaning and let it clean. All i ever really wanted was to disable WiFi on this thing so that it wasn't spamming my network with traffic or eating up a 2.4ghz channel with it's own AP signal.

Maybe one day I will return when I would like to have some of this other functionality that it looks like this solution might provide. BTW watching it's DNS requests from my PiHole this thing is nice and quiet now. :)

closed 2019-04-28 by Hypfer