Hypfer/Valetudo #182: Map lost after reboot

xoxys, 2019-04-22:
The map is completely lost in UI after every reboot of the device. Is this expected?

Using v0.3.0

cryptomilk, 2019-04-23:
Yes, `mirobo raw-command set_lab_status 1` needs to be run to tell the robot to not remove maps after a restart. I think we should add an option to Valetudo to enable it. Like Settings -> Maps to enable/disable lab mode or reset the map.

xoxys, 2019-04-23:
Thanks. That helps a lot. What about adding this to an FAQ in the meantime?

cryptomilk, 2019-04-23:
Feel free to do so, ask @Hypfer for edit rights.

Hypfer, 2019-04-23:
The Wiki is open 👍

xoxys, 2019-04-23:
Great!

xoxys, 2019-04-23:
Done

xoxys, 2019-04-23:
@cryptomilk @Hypfer 
Sadly `mirobo` commands does not work anymore. I've tried with `mirobo discover` and `mirobo raw-command` with no luck:

```
$ mirobo -d discover
INFO:miio.vacuum_cli:Debug mode active
INFO:miio.discovery:Discovering devices with mDNS, press any key to quit...
WARNING:zeroconf:Exception occurred:
Traceback (most recent call last):
  File "/home/srvmaster/roborock/lib64/python3.6/site-packages/zeroconf.py", line 2101, in send
    bytes_sent = s.sendto(packet, 0, (addr, port))
PermissionError: [Errno 1] Operation not permitted
```

cryptomilk, 2019-04-23:
You need to specify the ip and token! See `mirobo --help`

xoxys, 2019-04-23:
I've done this by ENV variables. Tried also with CLI options with the same result.

cryptomilk, 2019-04-23:
Works fine for me.

```
  krikkit  asn  (e) venv  ~  workspace  rockrobo  miio  mirobo status
State: Charging
Battery: 100 %
Fanspeed: 60 %
Cleaning since: 0:05:15
Cleaned area: 3.3375 m²
```

xoxys, 2019-04-23:
Strange... which firmware you are using?

xoxys, 2019-04-23:
And you are not in AP mode right?

cryptomilk, 2019-04-23:
v1748. However I don't think that matters, if you can ping the robot then probably the token is wrong. My robot is connected to my wifi network.

cryptomilk, 2019-04-23:
If you can understand `strace` output or wireshark those are you're friends :-)

xoxys, 2019-04-23:
I've used the token displayed in Valetudo UI. Robot is pingable and controllable through Valetudo UI

jowlo, 2019-04-23:
Just to chip in here, have you tried running your `mirobo` command as root?

The python error you pasted is an error in the zeroconf send function. I do not see why this should give an error, but i do not think this is in any way related to anything running on the robot.

xoxys, 2019-04-23:
@jowlo same

cryptomilk, 2019-04-23:
Your firewall might also block something ...

xoxys, 2019-04-23:
Yeah I've checked this too... Firewall is set to IPv4* any any...

xoxys, 2019-04-23:
no idea whats wrong...

jowlo, 2019-04-23:
Well, let's find out if your packet actually gets send:

 * Open terminal and run `tcpdump` on port `5353` (mDNS) and keep open:
```
sudo tcpdump -n port 5353
```
* Open a second terminal and execute your mirobo commands (as `root` to tackle one problem at a time).
```
sudo mirobo -d discover
```

`tcpdump` should show your packages originating from your IP to the IP of the robot - and possibly other mDNS traffic in your network.

xoxys, 2019-04-23:
@jowlo Hell.... it's some multicast fuckup with pfsense/unif switches and multiple vlans....

xoxys, 2019-04-23:
It is temporary working now.... Multicast is a piece of shit.

xoxys, 2019-04-23:
@jowlo Anyway, thanks for pointing me in this direction

xoxys, 2019-04-23:
I think I finally got it working. Last bastion was local iptables:

# allow mdns 
iptables -I INPUT -p udp --dport mdns -d 224.0.0.251 -j ACCEPT
iptables -I OUTPUT -p udp --dport mdns -d 224.0.0.251 -j ACCEPT

# allow other necessary port
sudo iptables -I OUTPUT -p udp --dport 54321 -j ACCEPT

Now connections with `mirobo` works again.

closed 2019-04-28 by Hypfer