Hypfer/Valetudo #183: which logfiles to look at? Some rooms are half or not any more cleaned

OhSoGood, 2019-04-24:
I regularly upgrade my S50 with latest FW and valetudo, and I've observed that these last 3-4 weeks 3 of my rooms got randomly poorly or not at all cleaned. I'm not speaking of what I see on the web map (I'll open another ticker for this) but of the dust which remains in the room. Sometimes, the same room is half cleaned, sometimes the S50 doesn't go at all in it (whereas there is no or few obstacles). I've visually checked the sensors which look fine.

I'd would like to investigate a little by myself but was wondering which logfiles to look at, or which command to run (either while cleaning or in a forensic a posteriori approach). Can anyone help me?

rand256, 2019-04-24:
AFAIK you should probably think of Valetudo as just a web interface to the vacuum, from where you can start or stop the default Roborock cleaning protocols which are always the same regardless whether you use Valetudo or not: they aren't (and seems can't be) modified anyway.

OhSoGood, 2019-04-24:
Understood. Can you point me to the log of the roborock?

Edit: last time I checked in /var/log, but forgot to look at 
/mnt/data/rockrobo/rrlog. Is this the right place?

rand256, 2019-04-24:
Yes, /mnt/data/rockrobo/rrlog contains mostly the processed and compressed output of rrlogd daemon.

Also some live logs are located at /dev/shm.

closed 2020-05-01 by Hypfer