Hypfer/Valetudo #185: Mistake in instructions

imbaimbaimba, 2019-04-24:
Hi!
Took me ages to figure out why my vacuum wouldn't flash, as I kept getting "No such command" for update firmware. Turns out there is a typo in the instructions:
`mirobo --ip 192.168.8.1 --token XXXXXXXXXXXXXXXX update-firmware image/output/v11_001768.fullos.pkg
`
Should be:
`mirobo --ip 192.168.8.1 --token XXXXXXXXXXXXXXXX update_firmware image/output/v11_001768.fullos.pkg
`

For the other Linux newbies out there, it might help to fix this :)

Hypfer, 2019-04-24:
Fixed

fvollmer, 2019-04-24:
I actually wrote this line. The doc at
https://python-miio.readthedocs.io/en/latest/vacuum.html#firmware-update
says "update-firmware" and I'm pretty sure I tested the line. I'll take
another look at it when I'm back home.

Hypfer, 2019-04-24:
hypfer@Valor:~$ mirobo update_firmware
Usage: mirobo [OPTIONS] COMMAND [ARGS]...
Try "mirobo --help" for help.

Error: No such command "update_firmware".
hypfer@Valor:~$ mirobo update-firmware
You have to give ip and token!

imbaimbaimba, 2019-04-24:
Hmm that is odd. After flashing now, I cannot access my robot on 192.168.8.1 either.

imbaimbaimba, 2019-04-24:
I previously had Valetudo running fine by a few versions older instructions. But one day, the robot couldn't find it's dock and ran out of power. When it turned on again, it reflashed itself to stock. So now I am just trying to install Valetudo again.

I just tried from the beginning again, by going step by step through the instructions. 

When I get to install python-miio I get a ton of errors ending in:
`Command "/home/pi/rockrobo/flasher/venv/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-0jqes7ln/cffi/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-6qaeelba-record/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/rockrobo/flasher/venv/include/site/python3.5/cffi" failed with error code 1 in /tmp/pip-build-0jqes7ln/cffi/`

I tried with sudo, and it didn't give me errors, so I assumed it worked. But I get this:
```
(venv) pi@raspberrypi:~/rockrobo/flasher $ mirobo update-firmware
Usage: mirobo [OPTIONS] COMMAND [ARGS]...

Error: No such command "update-firmware".
(venv) pi@raspberrypi:~/rockrobo/flasher $ mirobo --help
Usage: mirobo [OPTIONS] COMMAND [ARGS]...

  A tool to command Xiaomi Vacuum robot.

Options:
  --ip TEXT
  --token TEXT
  -d, --debug
  --id-file PATH
  --version       Show the version and exit.
  --help          Show this message and exit.

Commands:
  carpet_mode       Query or set the carpet mode.
  cleaning_history  Query the cleaning history.
  configure_wifi    Configure the wifi settings.
  consumables       Return consumables status.
  discover          Search for robots in the network.
  dnd               Query and adjust do-not-disturb mode.
  fanspeed          Query and adjust the fan speed.
  find              Find the robot.
  goto              Go to specific target.
  home              Return home.
  info              Return device information.
  install_sound     Install a sound.
  manual            Control the robot manually.
  map               Return the map token.
  pause             Pause cleaning.
  raw_command       Run a raw command.
  reset_consumable  Reset consumable state.
  serial_number     Query serial number.
  sound             Query and change sound settings.
  spot              Start spot cleaning.
  start             Start cleaning.
  status            Returns the state information.
  stop              Stop cleaning.
  timer             List and modify existing timers.
  timezone          Query or set the timezone.
  update_firmware   Update device firmware.
  update_status     Return update state and progress.
  zoned_clean       Clean zone.
```

I do not really know where to go from here. Anyone that can point me in the right direction?

imbaimbaimba, 2019-04-24:
I got no idea what this means, but it seems I needed to install these requirements:
`sudo apt-get install build-essential libssl-dev libffi-dev python-dev`
After that, I could install python-miio without any errors.
Now firmware-update worked, as opposed to firmware_update before. No idea how that makes any sense. I am running off a clean SD card on a Raspberry, no previously installed python-miio versions or anything.

Currently trying to flash the robot again to see if it works.

imbaimbaimba, 2019-04-24:
Updated succesfully, but 192.168.8.1 refused to connect :(

Hypfer, 2019-04-24:
Delete your build folder and start again from the beginning of the guide

imbaimbaimba, 2019-04-24:
> Delete your build folder and start again from the beginning of the guide

I just did. I will flash the SD card for the Raspberry clean and try again on a fresh Raspbian. Does it have something to do with the other open issue about unprovisioned robots not connecting?

I could maybe do this first this time?
`mirobo configure-wifi **SSID** **PASS** --timezone** COUNTRY/CITY**`

fvollmer, 2019-04-24:
AFAIK: The robot token can only be discovered if the robot is unprovisioned (or rooted or by extracting it from the xiaomi home app). I would suggest to use `configure-wifi` after rooting and installing valetudo.

If you are connected with multiple interfaces (e.g. via LAN to the internet and via WiFi to the robot) you may experience connection issues with miio. I guess is that the Multicast package is sent via the wrong interface.

rytilahti, 2019-04-24:
Wrt underscore vs. dashes in command names for mirobo, this depends on your click version (https://github.com/pallets/click/issues/1123). The new click versions (7+) will use the dash instead of underscore.

fvollmer, 2019-04-24:
@rytilahti Thanks for the info! I guess python-miio should specify `'click>=0.7'` in their setup.py.

imbaimbaimba, 2019-04-24:
> AFAIK: The robot token can only be discovered if the robot is unprovisioned (or rooted or by extracting it from the xiaomi home app). I would suggest not to use `configure-wifi` before rooting and installing.
> 
> If you are connected with multiple interfaces (e.g. via LAN to the internet and via WiFi to the robot) you may experience connection issues with miio. I guess is that the Multicast package is sent via the wrong interface.
> 
> Do you have any indications that the newly flashed firmware is active? Can you connect via ssh? @Hypfer Could it be that [3bcfb5c](https://github.com/Hypfer/Valetudo/commit/3bcfb5c74f3439c93235db49d9ae2be62a737afc) triggers the rollback again?

I couldn't connect via SSH. Trying to reflash now. However, now for some reason I cannot build the image. Doesn't seem to be my day :(

```
pi@raspberrypi:~/rockrobo/firmware/image $ sudo ../dustcloud/devices/xiaomi.vacuum/firmwarebuilder/imagebuilder.sh \
>      --firmware=../firmware/v11_001748.fullos.pkg \
>      --public-key=$HOME/.ssh/id_ed25519.pub \
>      --valetudo-path=../valetudo \
>      --disable-firmware-updates \
>      --ntpserver=192.168.178.1 \
>      --replace-adbd
sudo: ../dustcloud/devices/xiaomi.vacuum/firmwarebuilder/imagebuilder.sh: command not found

```

fvollmer, 2019-04-24:
@imbaimbaimba I guess you are at the wrong folder. You should be in `~/rockrobo/image/`

imbaimbaimba, 2019-04-24:
> @imbaimbaimba I guess you are in the wrong folder. You should be in `~/rockrobo/image/`

That worked. Couldn't see in the instructions that I had to go cd .. back to /rockrobo before creating image folder. :)
Fingers crossed that it works this time.

fvollmer, 2019-04-24:
I'm sure the instruction can be improved. I'm also kind of confused about the excessive use of `pushd` and `popd`. Step 4 could be

```
wget https://github ...
mkdir deployment
cd deployment
wget https://github.com/Hypfer/Valetudo/raw/master/deployment/valetudo.conf
mkdir etc
cd etc
wget https://github.com/Hypfer/Valetudo/raw/master/deployment/etc/hosts
wget https://github.com/Hypfer/Valetudo/raw/master/deployment/etc/rc.local
popd
```

and it appears to be like a `popd` is missing after "Download the latest firmware"

imbaimbaimba, 2019-04-24:
Now I'm getting `Error: Invalid value for "--token": Token length != 32 chars: 20` when trying to flash.

fvollmer, 2019-04-24:
Is your token only 20 chars long? This is strange. Try again to discover the robot and make sure you got the complete token.

imbaimbaimba, 2019-04-24:
Pressed the wifi reset button again, then I got a new token that was 32 characters. Weird.

imbaimbaimba, 2019-04-24:
192.168.8.1 refused to connect.
Tried SSH'ing again, but doesn't work like it used to. Gives me a long ECDSA key fingerprint - doesn't ask me for the password corresponding to my email address as how it worked last I had valetudo running (around 0.2)
Any clue if something is broken or if I am doing something wrong?

fvollmer, 2019-04-24:
The fingerprint of the robot changed. This should be normal.
Do you try to connect as root user? You need to do something like `ssh root@192.168.8.1`

imbaimbaimba, 2019-04-24:
Yep, exactly what I did. Maybe it’s not rooted and just reverted to stock fw?

fvollmer, 2019-04-24:
With the stock fw there shouldn't be an opened ssh port. Try `nmap 192.168.8.1`. What is the exact message you get with ssh?

imbaimbaimba, 2019-04-24:
My keys are in ~/.ssh. A reboot handled it, then it asked for the passphrase I set, and I am now SSH'ed in. So it must be rooted, but just nothing on port 80?

fvollmer, 2019-04-24:
This could mean that the unprovisioned bug is still unsolved. Try to start valetudo manually:
`/usr/local/bin/valetudo`. If this works you can use valetudo to connect to your WiFi.

imbaimbaimba, 2019-04-24:
Just as you wrote, I also found `service valetudo start` which worked.
Yeah it doesn't seem to autostart at least. Thank you so much for helping me through this, I apologize for my lack of skills to solve this myself.

Anything I need to know about going forward in relation to this? Will I keep loosing valetudo and have to start it myself? AFAIK the robot reboots every night.

fvollmer, 2019-04-24:
I'm not sure about the upstart problem. You could try to connect the robot to your WiFi.

imbaimbaimba, 2019-04-24:
Well it did connect to my wifi but it seems broken :(
https://i.imgur.com/dInvjaE.png
Just stays at loading state forever..

fvollmer, 2019-04-24:
try to reboot the robot and ctrl+f5 in your browser

imbaimbaimba, 2019-04-24:
Rebooting the robot took it back to AP mode. And valetudo did not start. SSH'ed in again, started it, and the web interface works. Then I typein my WI-FI information, it joins my wifi and I can connec to the IP but the same thing happens, just hanging.
start on net-device-up IFACE=wlan0 is in the valetudo.conf

imbaimbaimba, 2019-04-24:
Tried ctrl + F5, different browsers, incognito, from mobile etc. Same thing.

fvollmer, 2019-04-24:
I'm not sure how to explain this behaviour, but maybe someone else.
You could try to use `mirobo configure-wifi` and see if survives the reboot. Make sure to use the correct ip and token (you can use valetudo to show the token).

Otherwise I would suggest to wait for someone else with fresh ideas.

imbaimbaimba, 2019-04-25:
Tried setting WiFi with mirobo configure-wifi once before going to bed. Unfortunately, same result. 

The thread you referenced could be the same I guess. In case it occurs for them during their nightly reset. 

My case is similar to one of the replies in the sense that my Roborock never seen the Xiaomi cloud either. 

Thank you for trying nevertheless. Let’s hope someone has more ideas. 

If nothing comes up, I guess I will just eat my pride and install the Xiaomi app and run it stock. My wife is nagging me all the time about the damn thing not working 🤯

imbaimbaimba, 2019-05-03:
For whatever reason, after a couple of days my robot woke up with Valetudo now working. Haven’t touched anything in between.

So basically, robots that cannot be provisioned with Valetudo may actually work, but just need (a lot) of time to start working.

closed 2019-05-03 by imbaimbaimba