Hypfer/Valetudo #195: Consumable "Sensor" - Whats meant by that

balu-, 2019-04-27:
Thanks for this great software,
 I really like it, only one thing took me by surprise:
At the Consumalbes View there is a Consumable called Sensor, whats meant by that?
The other consumables i get, but what is this one?

jowlo, 2019-04-27:
The sensors are actually not "consumed", but should be cleaned on a regular basis. The same option is exposed in the original Xiaomi App, including instructions. I was asking myself the same question,  and came accross [this thread (in german)](https://www.roboter-forum.com/index.php?thread/20529-meldung-you-need-to-clean-the-sensor-now-was-ist-zu-tun/).

The thread links to two images from the official app, which are pretty self-explanatory. The idea is to just reset the counter after cleaning.
[image]
[image]

jowlo, 2019-04-28:
Would someone (@Hypfer maybe) close this? The issue section is quite messy already...

Hypfer, 2019-04-28:
@jowlo Mind adding a short explanation to the wiki?

closed 2019-04-28 by Hypfer

jowlo, 2019-04-28:
Done.