Hypfer/Valetudo #196: Latest working firmware version? (latest seems to be v11_001834)

HA-TB303, 2019-04-27:
Hi,

There is a version v11_001834.fullos.pkg now. Does anyone know what is the latest and greatest working version number?

BenRoe, 2019-04-27:
https://github.com/dgiese/dustcloud/wiki/Xiaomi-Vacuum-Firmware#tested-firmware

HA-TB303, 2019-04-27:
Thanks, but those last 3 additions on the dustcloud documentation came from me ;)

I wasn't planning on being the testing guinea pig al the time :)

rand256, 2019-04-28:
For now 1834 is latest of available firmwares as far as I see. However it writes way too many debug logs and wastes a lot of CPU for it. The same is true for 1832, 1830 and (a bit less for) 1826. The latest (by date) firmware without insane debugging is 1820.

cryptomilk, 2019-04-28:
v18xx is beta software! It has never been rolled out officially. The latest available firmware seems to be v1768.

rand256, 2019-04-28:
So beta firmware is not a firmware?

cryptomilk, 2019-04-28:
Well you can run it but don't expect it to be bug free ...

rand256, 2019-04-28:
I don't expect officially rolled out firmware being bug free too, obviously.

The question of this topic didn't mention beta or release channel of distribution, though, it was asked only which firmware is the latest, already referring to 1834.

HA-TB303, 2019-04-28:
Well, I wanted to know what is the latest WORKING version ;)

rand256, 2019-04-28:
Even 1834 is WORKING, though probably worse than it could.

I personally use 1820 for the last few weeks. Latest official release 1768 should also be fine.

HA-TB303, 2019-04-28:
Thx, I’ll stick to 1820 then.

closed 2019-04-28 by Hypfer