Hypfer/Valetudo #200: Large memory usage

oechslein, 2019-04-30:
No sure what happens on my version (directly build from github "Zoned_cleanup + Go_to via MQTT"/3b1a9449af6c9fbcf684c63f144e6b137c27167b):
[image]

And still increasing, I'm also getting "FATAL ERROR: CALL_AND_RETRY_LAST Allocation failed - JavaScript heap out of memory" from time to time.

[valetudo.log.zip](https://github.com/Hypfer/Valetudo/files/3130421/valetudo.log.zip)

The above screenshot was made right after a zoned cleanup.

oechslein, 2019-04-30:
Robo (Gen2 BTW) rebooted and uses now old official 0.3.1 version and continued cleaning (no zoned cleaning):
[image]

oechslein, 2019-04-30:
I disabled mqtt (set to false) and now I'm having normal/expected(?) memory consumption:
[image]

After that I reenabled mqtt (set to true) and memory consumption continously increases again:
[image]

jowlo, 2019-04-30:
Might this be a problem of the mqtt library, e.g. https://github.com/mqttjs/MQTT.js/issues/791 ?

Hypfer, 2019-04-30:
Might be. Last time I was hunting for memory leaks was before MQTT was added iirc

jowlo, 2019-04-30:
@oechslein could you try building valetudo with https://github.com/mqttjs/MQTT.js/commit/93046a51866d79fcfd8bd9021b5b4605d3e07f5a applied to the mqtt library? Seemed to have fixed the issue over there...

oechslein, 2019-04-30:
@jowlo Using  mqttjs/MQTT.js@93046a5 did not help:
[image]

jowlo, 2019-04-30:
Would have been too easy... Thanks for trying!

oechslein, 2019-04-30:
I wonder if I'm the only one seeing this?

oechslein, 2019-04-30:
See https://github.com/Hypfer/Valetudo/pull/201

Hypfer, 2019-04-30:
Even with mqtt turned off, this is still happening :( I'm experiencing OOM every few minutes

Hypfer, 2019-04-30:
```
Dummycloud is spoofing 203.0.113.1:8053 on 127.0.0.1:8053
Webserver running on port 80
Robot connected

<--- Last few GCs --->
biggest step 8.0 ms, walltime since start of marking 149 ms) finalize incremental marking via task GC in old space requ[19004:0x17e81f0]    85149 ms: Mark-sweep 17.3 (31.5) -> 9.2 (31.5) MB, 38.7 / 0.2 ms  (+ 114.5 ms in 119 steps since start of marking, biggest step 5.8 ms, walltime since start of marking 158 ms) finalize incremental marking via task GC in old space requ[19004:0x17e81f0]    86192 ms: Mark-sweep 17.0 (31.5) -> 9.4 (14.5) MB, 162.6 / 0.1 ms  low memory notification GC in old space requested

<--- JS stacktrace --->

==== JS stack trace =========================================

Security context: 0x5a6161f1 <JSObject>
    1: fromString(aka fromString) [buffer.js:314] [bytecode=0x2df56dad offset=164](this=0x3440417d <undefined>,string=0x3ae88199 <Very long string[288166]>,encoding=0x5a61ecf1 <String[4]: utf8>)
    2: from [buffer.js:177] [bytecode=0x2df56a5d offset=11](this=0x5d4aba61 <JSFunction Buffer (sfi = 0x5a6476ad)>,value=0x3ae88199 <Very long string[288166]>,encodingOrOffset=0x3440417d <undefined>,length=0...

FATAL ERROR: Committing semi space failed. Allocation failed - process out of memory
```

Hypfer, 2019-04-30:
Okay, this seems to be caused by a second (or third) map upload happening while there is already one being processed.

Hypfer, 2019-04-30:
Okay so my two fix attempts did reduce the amount of crashes per hour.

Still, it's not completely fixed.
Any ideas?

onovy, 2019-05-05:
tested current master (c1e7f5140cb292c38429bf7462ac55cbe4ae661d) with node 10 and full house cleanup (~ 50 mins). It looks like it's leaking much less than before, but I can still see some kind of memory leak. RES memory starts at "40MB" and before cleanup end it's at "75 MB" After finishing of cleanup it jumps down to "61 MB" . I'm not using mqtt (disabled) and I am watching map in Chrome browser for whole time of cleaning.

Hypfer, 2020-05-15:
stale issue

closed 2020-05-15 by Hypfer