Hypfer/Valetudo #206: Random firmware reset

xoxys, 2019-05-02:
Hi, im running valetudo 0.3.1 and today robo was not accessable, so i restarted the robot and after that he was back with default AP. No ssh connection possible und no valetudo available at port 80.

So maybe this is not fixed?

Hypfer, 2019-05-02:
There were some commits for the upstart config which could possibly fix this.

You might want to try those

xoxys, 2019-05-02:
How does the upstart fix Firmware resets?

xoxys, 2019-05-02:
Do you mean 6fd1be698a990e0746fa9eec460734fc8cdfd761?

Hypfer, 2019-05-02:
There is some kind of error counter somewhere in the roborock software which reverts to the previous firmware if it reaches a certain value.

I have no idea where it is and what causes it to increment, but I assume that going OOM is one thing that might do it.

https://github.com/Hypfer/Valetudo/commit/91088195f869d6d6f96bb612770e3bcc7a1d00b8 contains some mitigations against memory leakage which causes the player process to be killed.

xoxys, 2019-05-02:
Thanks for clarification :) I'll give it a try. First I have to re-flash the firmware to get back control over the robot...

xoxys, 2019-05-03:
Robot is back to life, lets see what happens.

xoxys, 2019-05-04:
This morning robot was not reachable and wlan led was off. So I decide to reboot, after that robot was reconnecting to wifi and NOT reseted. So there seems to be another problem. valetudo log contains:

```
Loading configuration file: /mnt/data/valetudo/config.json^M
Dummycloud is spoofing 203.0.113.1:8053 on 127.0.0.1:8053^M
Webserver running on port 80^M
events.js:183^M
      throw er; // Unhandled 'error' event^M
      ^^M
^M
Error: getaddrinfo EAI_AGAIN mymqtt.example.com:8883^M
    at Object._errnoException (util.js:992:11)^M
    at errnoException (dns.js:55:15)^M
    at GetAddrInfoReqWrap.onlookup [as oncomplete] (dns.js:92:26)^M
Loading configuration file: /mnt/data/valetudo/config.json^M
Dummycloud is spoofing 203.0.113.1:8053 on 127.0.0.1:8053^M
Webserver running on port 80^M
events.js:183^M
      throw er; // Unhandled 'error' event^M
      ^^M
^M
Error: getaddrinfo EAI_AGAIN mymqtt.example.com:8883^M
    at Object._errnoException (util.js:992:11)^M
    at errnoException (dns.js:55:15)^M
    at GetAddrInfoReqWrap.onlookup [as oncomplete] (dns.js:92:26)^M
```

and boot.log:
```
 * Stopping flush early job output to logs^[[74G[ OK ]^M
 * Starting configure virtual network devices^[[74G[ OK ]^M
 * Stopping System V initialisation compatibility^[[74G[ OK ]^M
 * Starting system logging daemon^[[74G[ OK ]^M
^M
dnsmasq: unknown interface wlan0^M

```

Hypfer, 2019-05-04:
`dnsmasq: unknown interface wlan0`

That sounds very broken

xoxys, 2019-05-04:
... Flashed with a fresh firmware yesterday. Works fine for a whole day, and works also after a reboot. No idea what's wrong

xoxys, 2019-05-05:
@Hypfer does the reboot command over ssh works for you? Running `reboot` from a SSH session kills my current session but does not bring back the robo until i hardly power it off with the physical button and turn on again

Hypfer, 2019-05-05:
Yup works here

xoxys, 2019-05-05:
Ok so what I've done:
I've build a fresh Firmware image from v1810 and v1820 both have a dnsmaq error in the Boot log also without adding valetudo to the image. So i decide do remove dnsmasq from upstart. After reboot all seems to work. AP mode and provisioning works fine. Also I notices sometimes a network error in my valetudo log. It seems that the upstart script does not respect the network-start directive... Maybe there is a small gap between "network interface is up" and network interface can reach my mqtt server. To fix this i've added a 30s sleep to pre-script section in the upstart config. 

For now, no more errors in valetudo log and reboot works also. Lets see what happens tomorrow :D

xoxys, 2019-05-08:
Robot is now up and running since 3 days, but: Lost the map today.....

tadly, 2019-05-08:
Because I had my robot also reset on my 2 days ago, I was searching for related issues.

For me, it took about a week before the robot reset itself (I think, didn't pay to much attention)

In regards to the error counter, could it be possible they have it in memory?
Was thinking that maybe restarting the player process once a day (cron) could be worth a shot until a real solution was found.
Unless you guys think I'm way off base :)

matthiasharrer, 2019-05-08:
There is already a memory limit with the upstart config on master which should prevent the resets I think

xoxys, 2019-05-08:
Currently for me its very hard to get a stable solution. For me the memory limit does not solve the reset issue finally.

matthiasharrer, 2019-05-08:
Do you use the memory limit already? Well then I am afraid I cannot help :( I do think however that restarting player is not the right direction to go torwards .. maybe just downgrade valetudo to a more stable version

xoxys, 2019-05-08:
Jep using the limit :) dont worry i know its hard develop for a "closed source" device. Random resets and my current map issue are hard to debug.

desq42, 2019-05-13:
…same here; memory limit set, reboot & firmware reset this night at 3:52 am. 
(My Gen1 "lives" in the bedroom. - my wife was not amused… ;-)

xoxys, 2019-05-13:
For me the robot is stabe since more than a week with my changes. The map was back after a reboot, no idea why the map wasn't displayed

tadly, 2019-05-13:
@xoxys would you mind telling me where exactly to remove dnsmasq from?

Also, what might be interesting is how many times the vacuum has been active.
I'd expect this to influence the time when the vacuum resets. (I'm actually down to 2x a week)

dugite-code, 2019-05-13:
@tadly @xoxys I had the error `dnsmasq: unknown interface wlan0` in my boot log as well modifying the option  bind-interfaces in `/etc/dnsmasq.conf` to bind-dynamic cleared that error in boot.log

So far this doesn't appear to have cased any issues, probably better than disabling dnsmasq altogether

xoxys, 2019-05-13:
@dugite-code good catch :) I'm not sure if `dnsmasq` is really required, so I decided to remove it from my setup. But yes you should be careful with these changes...

@tadly you should maybe follow @dugite-codeinstructions first, bu if you look for the file I think it is under /etc/upstart

rassaei, 2019-05-15:
Hi everyone, last night my robot also reset its firmware and woke us up in the middle of the night.
I didn't have much time today before work, so I just tried the same command to push the firmware to the robot I did last time (still had it in my history). It didn't work because the token was wrong. I started discovery and received a different token. With that new token it kinda worked but the robot told me to charge it before updating it, so I just left for work.
Is it possible that the token changed? I thought it always stays the same

posixx, 2019-05-15:
i also have the same problem; after some days the rockrobo looses config. After rebooting the rockrobo i can connect to the internal wifi, but no ssh or GUI access. So it seems it is completely reset. Very annoying. Hope this is solved soon..

xoxys, 2019-05-19:
@posixx It's not that simple. If you have any suggestions please share with us :)

xoxys, 2019-05-19:
@rassaei As I know the token will be randomly created at first boot. So after a full reset the token will be re-generated.

dugite-code, 2019-05-20:
@posixx @xoxys Found something new: On the German roboter-forum.com there was a thread suggesting they stopped seeing this issue once they added the missing `/mnt/default/roborock.conf` file. If you are missing that as well you can create one following the [CCC to CE conversion](https://github.com/dgiese/dustcloud/wiki/Vacuum-Robots-CCC-to-CE-conversion#step-by-step-conversion-from-ccc-to-ce) guide

josch, 2019-05-20:
After every reset I'm flashing the same firmware image (based on 1792) on my gen2. That image contains `/mnt/default/roborock.conf`:

```
root@rockrobo:~# cat /mnt/default/roborock.conf
language=en
name=custom_A.03.0005_CE
bom=A.03.0005
location=de
wifiplan=
timezone=Europe/Berlin
logserver=awsde0.fds.api.xiaomi.com
```

But I'm still seeing reboots once in a while. I did not modify that image otherwise, so maybe multiple conditions have to be met.

cryptomilk, 2019-05-20:
Did someone test with PR 213 ????

xoxys, 2019-05-20:
Nope. If someone merge and tag a pre-/beta-release it would be easier to test because we dont need to build it locally

XCame, 2019-05-20:
Came home from a weekend trip yesterday, to find my Gen1 Xiaomi Mi Robot resetted and unprovisioned. I think with the original firmware because my language pack wasn't in english before I left.

Any chance I can provide you some useful information before I reflash it? 

I wouldn't mind opening it, if soldering for a serial connection would be necassary to gather information.

rassaei, 2019-05-23:
@xoxys thx, didn't know that!
@cryptomilk I built from master a couple of days ago and am testing that. Didn't notice the PR, needed the robo to run again :D Will try the PR when I have time again

Hypfer, 2019-05-28:
#218 breaks the map view updates

tobiaskohlbau, 2019-06-03:
I would like to add a few informations I gathered.

> There is some kind of error counter somewhere in the roborock software which reverts to the previous firmware if it reaches a certain value.
> 
> I have no idea where it is and what causes it to increment, but I assume that going OOM is one thing that might do it.
> 
> [9108819](https://github.com/Hypfer/Valetudo/commit/91088195f869d6d6f96bb612770e3bcc7a1d00b8) contains some mitigations against memory leakage which causes the player process to be killed.

You're right there exists a error count which is located under `/opt/rockrobo/watchdog/failure_count`. The file is created and managed by the watchdog process which is represented by the binary `/opt/rockrobo/watchdog/WatchDoge`. The related config file `/opt/rockrobo/watchdog/rrwatchdoge.conf` states:
```
# crash detect, if robot did not crash after "CRASH_IGNORE_MIN" minutes, then clear the crash counter.
env CRASH_IGNORE_MIN=600
```
The files `ProcessList.conf` and `ProcessListMT.conf` state the processes tracked by the watchdog. The `WatchDoge` binary has the expression:
```
9 < WATCHDOG_COUNT
```
so the magic value is 10 failures within 1 hour (otherwise the crash_ignore_min would take action).

In advance the `WatchDoge` has the ability to switch to the other partition as there exists the string `need to switch to another partition, wait hw watchdog timeout...` within the binary. It also look like the watchdog sets some sort flag onto the partition for knowledge if the partition is good and if both partitions fail the reset is executed.

IMHO the best solution would be to search which applications fails and find the root cause and fix it. Maybe it's caused by valetudo maybe it's a bad firmware I don't know.

roflcoopter, 2019-06-03:
I can confirm the above.
I set CRASH_IGNORE_MIN=1 and rebooted and have not had a reset in 4 weeks.
I have not noticed any other issues either

timbru31, 2019-06-12:
Today my gen2 decided to reset itself, too. (second time since April 2019 - so the reset occurs roughly every 4 weeks).
I was on FW 1748, now I've flashed to 1784 and changed the CRASH_IGNORE_MIN value, too.

@tobiaskohlbau have you made any progress with keeping the log files before the reset?

magnulu, 2019-06-13:
@timbru31: I am pretty sure my gen2 also reset itself yesterday. Maybe just a coincidence, but makes me think this could be triggered by something static/external?

tobiaskohlbau, 2019-06-14:
@timbru31 unfortunately I hadn't enough time to fiddle around with the robot lately. I hope I can find some time this weekend.

fvollmer, 2019-06-14:
I also noted that my robot did a factory reset yesterday. This is the first time it ever did this.

Hypfer, 2019-06-18:
I just noticed a factory reset as well. It happend just in time with the daily scheduled reboot so I guess that this might be the new cause since the memory leaks should not be causing any more issues

Poeschl, 2019-06-18:
Running mine with the adjusted `CRASH_IGNORE_MIN` flag for nearly 2 weeks now and now reset occurred so far.

ministryofsillywalks, 2019-06-22:
Just had a reset aswell on my Gen1 with 0.3.1.
I noticed my robot wasn't going on its cleaning runs so I tried to connect however I got no connection.
Opening the lid had my robot speaking chinese with me and looking at the wifi networks available its back to AP mode.
Guess I'll reflash again and try the crash_ignore_min change.

rand256, 2019-06-22:
@ministryofsillywalks in 0.3.1 resets may happen due to memory leakage and code issues in valetudo itself so you should definitely update to something newer. With current valetudo device sometimes resets too, but this happens rarer and the exact reason for that is unknown for now. Maybe it's not getting some special response (or requests?) from dummycloud so the firmware decides that something goes wrong and resets itself just because of that, or something alike...

xoxys, 2019-06-22:
Something newer than 0.3.1? I cant see any new release since 0.3.1...

rand256, 2019-06-22:
@xoxys sure, thus I'm not talking about new releases. You could probably [take a look](https://github.com/Hypfer/Valetudo/commits/master) on a number of new commits published after 0.3.1 release to see what is new.

xoxys, 2019-06-22:
Sorry guys but I say it again... If there are important Bugfixes or stability issues which are fixed in master branch so please tag a new minor release! It makes it so much easier for everyone to upgrade.

ministryofsillywalks, 2019-06-22:
> @xoxys sure, thus I'm not talking about new releases. You could probably [take a look](https://github.com/Hypfer/Valetudo/commits/master) on a number of new commits published after 0.3.1 release to see what is new.

How do I go about updating my robot to include these commits?

xoxys, 2019-06-22:
@ministryofsillywalks you have to build it by your own. You can find some build instructions here https://github.com/Hypfer/Valetudo/blob/master/develop/HOWTO.md

ministryofsillywalks, 2019-06-22:
> @ministryofsillywalks you have to build it by your own. You can find some build instructions here https://github.com/Hypfer/Valetudo/blob/master/develop/HOWTO.md

thanks! I will look into it, but I would also love to see new proper releases if serious bugs are fixed.

Skeletorjus, 2019-06-22:
> > @xoxys sure, thus I'm not talking about new releases. You could probably [take a look](https://github.com/Hypfer/Valetudo/commits/master) on a number of new commits published after 0.3.1 release to see what is new.
> 
> How do I go about updating my robot to include these commits?

If you take a look at #266 I've created a Dockerfile which will compile it for you.

balu-, 2019-07-03:
Any news here?

My vacuum has reseted it's self again (i noticed today)
So at least for me 

>        env CRASH_IGNORE_MIN=600

did not work.

deflomu, 2019-07-03:
Isn‘t 600 the default value? I think you should set it to 1 to prevent the robot from resetting ( meaning it will only reset if there are at least 10 errors in 1 minute instead of resetting if there are 10 errors in 600 minutes)

balu-, 2019-07-03:
> Isn‘t 600 the default value? I think you should set it to 1 to prevent the robot from resetting ( meaning it will only reset if there are at least 10 errors in 1 minute instead of resetting if there are 10 errors in 600 minutes)

Sorry i was not specific enough i set the value to 10 
(to not disable this feature completly but to make it very unlikely that in normal usage 10 errors accumulate). 

To add further details, 
I've a gen2 device and was using Valetudo Beta 0.3.1 with the firmeware 1782. 
The robo is in a wifi thats not connected to the internet.

I guess my next step will be trying the newest firmeware and building Valetudo from Master branch

desq42, 2019-07-09:
…build newest firmware last week. Yesterday eve my robot gen1 flashed itself back to the stock firmware. Factory-Reset. New token yadda, yadda. Is this issue under scrutiny?

tadly, 2019-07-09:
huh...
So last time mine reset I did set `CRASH_IGNORE_MIN=1` and because of @desq42 comment I wanted to check the uptime of my robot (Should be around 4 weeks or something).
```sh
$ uptime -p
up 5 hours, 41 minutes
```
But that's not all. `CRASH_IGNORE_MIN` was back at `600`.

In other words, it is very likely that my robo would have fully reset itself with the next _cycle_ (that's what I'm going to call the repeated reboot/reset now).

josch, 2019-07-09:
I can confirm the observation of @tadly. I also had set `CRASH_IGNORE_MIN=1` to check whether this is a possible workaround. We have the robot in our bedroom so it's not possible that we would miss any robot reset at 3:00 AM in the night. The firmware reset which always had woken us up because of the repeated noises did not occur yet. But when I checked now, `CRASH_IGNORE_MIN` was indeed back at `600` so something was rewriting that file...

tadly, 2019-07-09:
`chmod 444 /opt/rockrobo/watchdog/rrwatchdoge.conf`
Let's see if this helps. (Makes the file read-only for all users)

I also cobbled together a quick&dirty script to poll the vacuum and post a system notification on my main PC (using cron) in case that value changes again.
Let's see how long it takes this time...

```sh
#!/usr/bin/env bash

HOST="vacuum"
EXPECTED="env CRASH_IGNORE_MIN=1"

conf=$(scp ${HOST}:/opt/rockrobo/watchdog/rrwatchdoge.conf /dev/stdout \
    | grep 'CRASH_IGNORE_MIN=')

if [ $? -ne 0 ]; then
    # Not reachable. I have my pc with me at work.
    exit
fi

if [ "${conf}" != "${EXPECTED}" ]; then
    notify-send -u critical "Vacuum config changed" "$conf"
fi
```

desq42, 2019-07-09:
…this seems to be an issue with all rooted fw… Dustcloud seems to have the same issue:
[https://github.com/dgiese/dustcloud/issues/206](url)
I´m hesitating to reflash dustcloud&Valetudo again. 
(every 2 Weeks the token hazzle and deployment.) 
@tadly could you report, if your rrwatchdoge.conf despite 444?

tadly, 2019-07-09:
@desq42 you bet I will. At least that was the plan :)

[311 more replies]

closed 2020-06-30 by Hypfer