Hypfer/Valetudo #211: Map View still does not work through a Reverse Proxy with basic auth

DavidB81, 2019-05-04:
Hi,
I updated with valetudo 0.3.1 and Nginx in connection http (80). The map display in real time still does not work.
Same problem with version 0.3.0.

would it be possible to update valetudo differently because each update must free space on Roborock and reinstall then configure Nginx?
Best regards

fvollmer, 2019-05-04:
I'm using a valetudo 0.3.1 with a nginx reverse proxy for basic authentication (my nginx is not running on the robot). I don't have any map related problems. I posted some details here: https://github.com/Hypfer/Valetudo/wiki/FAQ#how-can-i-add-password-authentication, maybe they are useful for you.

DavidB81, 2019-05-05:
I specify that I had to free space and that Nginx reverse proxy is directly running on roborock. 
I have to do this after each update of Valetudo. It's a bit heavy to implement.

What kind of hardware is Nginx reverse proxy running?

fvollmer, 2019-05-05:
Might be a stupid question, but does your map work without the reverse proxy?

I would suggest you to open the developer console of your web browser and look for errors. There should be an error which indicates the problem. Feel free to post them here.

How does your configuration look like? I guess valetudo listens only to localhost on a custom port and the nginx to reverse proxy to listen on port 80? If I read https://github.com/Hypfer/Valetudo/blob/a9fa64fcec23639b7367a0d3fa61b84cd07eaebe/lib/Valetudo.js#L23 correctly you need extra configuration if you don't run valetudo on prt 80. Keep in mind that need to proxy http and web sockets. How does your nginx configuration look like?

DavidB81, 2019-05-05:
The real-time map work correctly without Nginx. With Nginx i must refresh the page of web browser to show the progress of the map.

Nginx listen by domain name and custom port for reverse proxy to valetudo localhost on port 80.
Valetudo listen only to localhost on port 80.

DavidB81, 2019-05-05:
It's my config:
server {
    listen XXXX; port number
    server_name monroborock.mamaison.com; domain name

    location / {
        auth_basic \"Restricted Content\";
        auth_basic_user_file /etc/nginx/.htpasswd;
        proxy_pass http://127.0.0.1:80;
    }

fvollmer, 2019-05-05:
I'm pretty sure you need to set the headers correctly and you need websockets. I would guess something like this.

```
server {
    listen XXXX; port number
    server_name monroborock.mamaison.com; domain name

    location / {
      proxy_set_header        Authorization "";
      auth_basic              "Restricted Content";
      auth_basic_user_file    /etc/nginx/.htpasswd;
      proxy_http_version      1.1;
      proxy_set_header        X-Real-IP $remote_addr;
      proxy_set_header        X-Forwarded-For $proxy_add_x_forwarded_for;
      proxy_set_header        X-Forwarded-Proto $scheme;
      proxy_set_header        Host $host;
      proxy_set_header        Accept-Encoding "";
      proxy_set_header        Upgrade $http_upgrade;
      proxy_set_header        Connection "upgrade";
      proxy_pass              http://127.0.0.1:80; # the ip address of the robot
      proxy_read_timeout      90;
    }
}
```

I just plugged your data into the example I posted earlier, but I didn't test it.

DavidB81, 2019-05-05:
Thank you so much. it works perfectly with your configuration example.
You were right the headers were not complete.

You can close the subject.
best regards

closed 2019-05-05 by Hypfer