Hypfer/Valetudo #214: Zoned cleanup touch targets nearly impossible to hit on mobile

cnrd, 2019-05-07:
Hi

I'm on Valetudo 0.3.1

iPhone XS Max (iOS 12.2)

The red "move" touch target on the map, when placing zones to clean up is nearly impossible to hit in safari on iOS.

Also it seems that I am unable to drag the red target in Chrome on desktop, if I hit it outside the square, while it works fine inside the square.

My guess is that the red target is only 1 px in size?

matthiasharrer, 2019-05-08:
Hi,

the red "button" is actually a delete button (on tap) and not a move button.
You can move the zone grabbing anywhere on the zone.

It would be nice to have icons in these buttons so that this becomes clearer ..

cnrd, 2019-05-08:
Oh, thanks I hadn't realised that :-) Icons would probably be good.

matthiasharrer, 2019-05-08:
:+1: Can this issue be closed?