Hypfer/Valetudo #215: Vacuum not moving on map

cnrd, 2019-05-07:
Hi

Valetudo 0.3.1
Chrome and iOS Safari (iOS 12.2)

The vacuum is not moving on the map while cleaning, the line is still being drawn. (The vacuum drawing is turning on the spot at the dock).

[image]

Hypfer, 2019-07-15:
Is this still happening?

VirtualGit, 2019-08-29:
Hi, I don't know if it is related, but I experiment the same issue through an apache proxy. 
Here is my conf :

I have a public dns pointing my public IP. An apache server is serving the robot web UI through a simple proxy configuration :

```
ProxyPass /robot http://192.168.xxx.xxx
ProxyPassReverse /robot http://192.168.xxx.xxx
```
(192.168.xxx.xxx si the robot IP)
I access the robot UI through : http://mydns.net/robot

The UI seems globally ok, except the realtime map update.

I guess the issue may comes from a websocket trying to connect to the root server path, in vacuum-map.js :
`ws = new WebSocket(`${protocol}://${window.location.host}/`);`

Here is two screenshots of firefox.
From http://192.168.xxx.xxx/ (no problem, mas is updated in real-time) :
[image]
From http://mydns.net/robot/ (problem, map is not updated) :
[image]

Switching screen to home, and back to map, update the map (lines) but the robot still don't move. 

Valetudo 0.4.0

acht, 2019-10-31:
I have the same issue with Valetudo 0.4.0 whenever the network latency increases. In the local network it works fine from remote the map update stops after some seconds or do not update at all.

Hypfer, 2020-05-01:
Closing this because it's very old.

If it is still happening please open a new issue

closed 2020-05-01 by Hypfer