Hypfer/Valetudo #216: Map generation issue/question

cherokee93, 2019-05-07:
[image]
Hello!
I'm new, I have the robot since yesterday and don't really know where to find the answer.

The above is my map, any idea why it is not generating? Robot is cleaning right now, but no map is being generated. On Xiaomi App I can see the map (no hosts entries) and everything.
Maybe something related to the map png feature that got removed?
I'm sorry if it's not an issue

Thanks for this software, love Valetudo ❤️

Hypfer, 2019-05-07:
**(no hosts entries)**

cherokee93, 2019-05-07:
🤦‍♂ I removed host entries because I couldn't use Valetudo, it loaded and loaded and loaded. 

Ok, i'll re-add hosts entries and retry then.  🤣 
Sorry for bothering. Thanks

cherokee93, 2019-05-07:
Ok now Valetudo  works perfectly ✔️ 
But (rightfully) Xiaomi app is not working. Is there any way to make it work? Or is something I need to give up?

rand256, 2019-05-07:
Yes, you need to choose whether you use Valetudo OR Xiaomi app, not both.

closed 2019-05-07 by Hypfer