Hypfer/Valetudo #220: Select multiple zones

Luftloch80, 2019-05-09:


Luftloch80, 2019-05-09:
Is it possible to select multiple Zones for cleaning within the webinterface?
I have Single Zones for kitchen living room and so on. 
I Want to clean several Zones from the Web Ui without creating Zones for each possible configuration 
Thx

BenRoe, 2019-05-16:
You can check multiple zones in the zone tab and then press the clean zone button.

rand256, 2019-05-26:
As far as I see, there's no way to "check" zones in the zone tab, you can only edit them from there, one by one.

hal295, 2019-07-20:
Hi, from Valetudo version 0.4.0 isnt possible to select multiple zones from zones tab like in prev version.
Please can somebody make this feature again? I am cleaning often several rooms/zones at time. Thanks!

ppisljar, 2019-09-22:
i'll probably take a look at this

Markus-, 2019-12-19:
Anything new?

rand256, 2019-12-19:
<details>
  <summary>This is how it looks in Valetudo RE</summary>

[image]

</details>

closed 2020-06-09 by Hypfer