Hypfer/Valetudo #225: Fullscreen browser on android to display valetudo?

momo42, 2019-05-11:
Hello together,
first thanks a lot for all your work! I've installed Valetudo on my v1 and would like to use my Nexus 4 phone to control the robot. Using Firefox (Android) I managed to create a start screen icon (with nice valetudo favicon). So far so good.

The problem is that I was not able to find a way to make Firefox display the Valetudo site in real fullscreen. I always have the annoying address bar on top. This leads to a scrolling 'home' page. And even more annoying I'm not able to set new timers because the buttons of the pop up menu beeing cut off.

I know Nexus 4 screen is on the very small side today but thanks to LineageOS it still gets weekly Android 8.1 updates. So it is still my daily driver.

I don't ask you to go even more 'responsive'. I think if I'd be able to use the full screen size of my phone this would be enought to display valetudo correctly. Is there anyone who managed to get valetudo to be displayed in real fullscreen on android?

momo42, 2019-05-11:
perhaps there is even some open source component to display web apps like Valetudo as native android app? Just thinking of a minimal fullscreen browser with hard coded url. I'm not a developer but I know that the app of my bank in fact is a fullscreen browser to display the responsive banking website.

jowlo, 2019-05-11:
It does not work for me in Firefox or Lineage's Browser either. It does work in Chrome on Android, though.

This feature is [actually implemented](https://github.com/Hypfer/Valetudo/blob/master/client/manifest.json#L12). Setting `"display" = "standalone"` should take care of hiding browser UI elements when you start it via the home screen launcher. You could try setting this value to `fullscreen`, however, `standalone` is the correct way to do it according to [the documentation](https://w3c.github.io/manifest/#display-modes).

momo42, 2019-05-13:
Thanks for this explanation. This is sad news. Although using an Android phone I'm not a big fan of Google. So installing Chrome on my (intentional) Google-Play-less Nexus 4 is not an option. :-/
Any further ideas?

jowlo, 2019-05-13:
Well, apparently, Firefox only enables the 'Add to homescreen' option for sites served over HTTPS, see https://developer.mozilla.org/en-US/docs/Web/Progressive_web_apps/Add_to_home_screen#How_do_you_make_an_app_A2HS-ready .

You could set up a reverse proxy with SSL in some other device and connect via that. I did not test this, though.

momo42, 2019-05-13:
Hmm. This is interesting. In fact I had no problem to add the (http) valetudo site to my homescreen via Firefox 'Add to homescreen' function!

The only problem is that I cannot get rid of this silly browser bar on top. I searched the net a bit and also found other people complaining about this exact problem with other websites. The url bar vanishes when you scroll the page down x pixels but nowadays a lot of mobile websites have their own static menu and some scrolling 'frame' underneath. And this prevents firefox from hiding its url bar. m(

Some say this being a security 'feature' to prevent malicious sites from hiding the browser ui and replacing it with some scam. Makes sense to me but in this case I would expect some toggle to switch that off.

But I want to say that I'm not fixed to Firefox. Any open source solution for this problem is welcome. :-)

jowlo, 2019-05-13:
> This is interesting. In fact I had no troubles to add the (http) valetudo site to my homescreen via Firefox 'Add to homescreen' function!

There are two ways of doing this. You can always add a shortcut to your homescreen from somewhere in firefox's menu, however, for sites correctly set up for PWA, firefox shows an additional icon right in the URL bar. It's a house with a plus sign in it... Check, for example https://yr.no.

Only when added via this button, URL bar is hidden for pages correctly set up. Did you try Valetudo through an ssl proxy?

jowlo, 2019-05-13:
Apparently, you can enable `dom.serviceWorkers.testing.enabled` in `about:config` to overwrite the HTTPS restriction on ServiceWorkers, and possibly also for the whole PWA , see [here](https://github.com/w3c/ServiceWorker/issues/658#issuecomment-87283965).

momo42, 2019-05-13:
Ok, thanks for your time! I've also read the linked discussions. Here are my results:

- the demo page works as expected (house icon with plus shows up on top. When added to home screen  via this icon, the demo page behaves like a standalone app.
- I'm also able to add the demo page to my home screen via menu item (AND via long pressing the url bar). The resulting icon looks the same but now it behaves like a normal webpage and gets opened in a browser tab with url bar on top etc.
- Valetudo site of my vacuum has no house+ icon. So I'm only able to add a 'shortcut' on the home screen.
- Switching the `dom.serviceWorkers.testing.enabled` option doesn't change anything for me.
- I have no idea of setting up ssl proxies 
:-(

Cooper81, 2019-05-20:
I Just use fully kiosk browser for fullscreen browsing ;)

closed 2019-06-04 by Hypfer

momo42, 2019-06-06:
Just asking. Why is this request marked as invalid? In my opinion being able to display Valetudo like a native Android app using open source apps is a valid goal. Fully Kiosk browser seems to be a proprietary closed source app with paid extra features. Is this really the only way to have valetudo in fullscreen? If not, please tell me more! Thanks for your time!

Hypfer, 2019-06-07:
https://github.com/Hypfer/Valetudo/blob/master/client/manifest.json#L12

Valetudo already implements the [Web App Manifest](https://developers.google.com/web/fundamentals/web-app-manifest/) since 0.0.6. You just need any browser which properly supports this to use the "Add to Homescreen" option. I'm using chrome, safari on iOS also works.

momo42, 2019-06-08:
Thanks. :-) Will try to get Chromium onto my device.

Would be really nice to have some kind of 'compatibility matrix' in wiki. Perhaps other users know tricks to make it run fullscreen on other mobile browsers too.

At least Firefox (mobile) 67.0 and Jelly 7.1.2 (LineageOS stock browser) don't support valetudos Web App implementation at this time, while they do works for other websites.