Hypfer/Valetudo #230: add the option of https via self signed certificate

momo42, 2019-05-13:
I'd like to add the Valetudo site of my Robot to my android homescreen via firefox 'add to homescreen' function (house icon with plus in it). Here https://github.com/Hypfer/Valetudo/issues/225 I was told that Firefox requires a https connection to display this option.

Aside from that encrypted access is always better than cleartext - even in home networks. Nowadays you never know if a nice IoT toy is spying on you. It would be really nice to have this option available at some point in the future.

Poeschl, 2019-06-04:
Implemented a ssl config for own certificates in my fork, working very well except websocket requests.
Going on...

Hypfer, 2021-02-16:
Since there's unfortunately no way to get this working without either installing your own root certificates everywhere or doing some arcane letsencrypt wildcard DNS hacks, I'll close this for now.

The recommended approach to do this would be to throw a reverse proxy in front of Valetudo or tell the firefox devs that requiring ssl for `add to homescreen` makes absolutely no sense in a local network since there is no non-hack way to have valid SSL there.

closed 2021-02-16 by Hypfer