Hypfer/Valetudo #235: error state

dh-harald, 2019-05-16:
Hi,

I'm using the latest version 0.3.1 Valetudo (base: 001748)...
Today, my robot stucked in a carpet.
In homeassistant, the state remains Cleaning. When I went to the web ui, the state was discharging (if I remember well), but when I pressed the start button, it turned into undefined.
I think, there's something with error handling... (same behaviour after reboot)

My robot will remain this state until evening (I'm not at home), so I can debug it for you.

stich86, 2019-05-17:
i'm asking the same thing.. using the original Mi Cloud, if the vacuum as an error (wheel suspended, clean dust bin, laser error and so on) i receive a notification on my phone. Is it possible to get these events via MQTT and\or HA and send a notification using the companion app?

thanks!

hanzoh, 2019-05-18:
Weird that three people are coming up with this in 2 days. Mine had its main brush blocked, but HA said nothing about errors and the web UI first gave Undefined and later Charger disconnected.

Getting the errors via MQTT would be great.