Hypfer/Valetudo #237: Zone Cleanup is Vertically Flipped

dlashua, 2019-05-16:
When using zoned cleanup via MQTT:

/valetudo/rockrobo/custom_command
{
  "command": "zoned_cleanup",
  "zone_ids": [
    "kitchen"
  ]
}

Occasionally the area to be cleaned will be vertically flipped on an axis with the dock. In other words, if there is an area of my house below (on the map) the dock, the zoned cleanup will send the vacuum to an area the same shape and horizontal location as intended, but above the dock instead.

This doesn't seem to happen if I use the Zone selector from within Valetudo and pick the zone. Additionally, I can stop the zoned_cleanup activity and then restart it using MQTT and it will then pick the correct area.

Firmware 3.3.9
Build 001748
Valetudo 0.3.1 (built from latest git repo)
Gen 2 Roborock S5

matthiasharrer, 2019-05-16:
@aprosvetova This is still the known issue with the mutable config or should this be fixed?

aprosvetova, 2019-05-16:
@matthiasharrer it's known but still not fixed. @Hypfer wanted to refactor Vacuum.js code so it makes up a new object with coords instead of using config object directly.

dlashua, 2019-05-30:
Since this commit ( https://github.com/Hypfer/Valetudo/commit/de0a28db98d577746ca7bcc185083035dafbdeb1 ) I have not seen zones vertically flipped any more. It's only been a few days and I've not used the feature much (since the vertically flipped clean up doesn't really do what it's supposed to). But I'll be ramping up my usage of this feature again to see if this resolves the issue.

dlashua, 2019-05-31:
I was overly hopeful. This is still an issue.

Hypfer, 2019-05-31:
Are you sure about that? Do the zones in the configuration file change?

dlashua, 2019-05-31:
The zones in the configuration file don't seem to change... but... here's what I did...

Yesterday, downloaded new Valetudo source from git, built, and installed on vacuum.
Added 3 zones for testing.
Tested and they worked fine multiple times.

Today, did nothing but go to "Zones" and choose my zone. The zone was vertically flipped resulting in my vacuum attempting to clean an area that is not even accessible and is outside of my house.

Hypfer, 2019-05-31:
So there was a restart between those cleanups?

Hypfer, 2019-05-31:
Can't reproduce. :/

Also, I don't see anything in the code that could still be causing this issue.

dlashua, 2019-06-02:
I don't remember a restart, but it's possible. Does a restart have the potential of changing coordinates?

Hypfer, 2019-07-14:
Is this still happening? I wasn't able to reproduce it

closed 2019-10-26 by Hypfer