Hypfer/Valetudo #243: Feature Request: Current Zone

dlashua, 2019-05-30:
Since we know the location of the vacuum (though this isn't exposed in MQTT) and we know the coordinates of each zone, it seems like we should be able to publish a "current_zone" state that indicates which of the zones the vacuum is currently in. This would be useful for finding the vacuum when it has an error, and also knowing which zones got cleaned during a "full cleaning" cycle. It's also useful for knowing if the vacuum made it to the requested zone.

Hypfer, 2020-05-01:
The new section feature also does this 🤔

ne0phyte, 2020-06-09:
I started working on an api method for this to show the current zone on the home screen. Haven't looked into the MQTT code yet, but I guess that shouldn't be too hard to add?

For now it fetches the zone(s) the robot is currently in and returns zone id and name.
Since zones can overlap or you might create one for the whole room and one for smaller subsections I'm not sure what would be the best way to handle that.

Right now it simply returns the matching zones in the order the zones are configured.
Maybe it would be good to sort the matches by zone size and always return the smallest matching zone first?

dmlemos, 2021-05-31:
Isn't this already covered on the latest releases?

closed 2021-08-14 by Hypfer