Hypfer/Valetudo #258: Reboot at 3am. Lack of infos.

davekr8, 2019-06-08:
Hey there,
I am almost sure that there is a scheduled reboot after 3am (for robots of 1st and 2nd gen.). I have both types of robots and they seem to perform some logging/zipping actions and trigger a reboot between 3:15 and 3:25 in the morning. As my miio2mqtt python app loose connection in the period of time I was asking myself:

*Is there are config to adjust the reboot time or even disable the scheduled reboot?*
*Is the watchdog triggering the reboot, or any other provess?
*If the watchdog is the origin of the reboots, is there any sources for the programm?

I did some days googeling around with minor succes... thus, I place my questions here - in hope for some input or resonance. Maybe you have the same questions in mind.

dugite-code, 2019-06-11:
Yes the robot reboots every day between 3:30am and 4am the only config file I know of that mentions reboots is `/opt/rockrobo/watchdog/rrwatchdoge.conf` But I have never seen any explanation on what these options do. Additionally you can always install `molly-guard` and forcibly prevent reboots.

That said as @Hypfer has stated before it's probably a bad idea to intentionally break reboots https://github.com/Hypfer/Valetudo/issues/80#issuecomment-457127944 Why do you want to prevent the reboot?

closed 2019-06-11 by Hypfer

davekr8, 2019-06-11:
what about "help wanted" label? no interest in sharing knowledge?

Hypfer, 2019-06-11:
It's not Valetudo specific so it should be moved to https://github.com/dgiese/dustcloud/

davekr8, 2019-06-11:
valid argument!