Hypfer/Valetudo #259: valetudo can't get device state after boot

rand256, 2019-06-09:
I've seen several times the next issue with latest master branch on my Gen2 device:

After I booted the robot (from fully offline state), the valetudo got loaded, I entered its site and saw the map (since it's persistent) *but* by some reason the device status on the "home" tab never finished loading. So it couldn't show the robot's state, battery level etc. Therefore I was unable to launch the cleaning or do almost anything since the state was unknown.

Next I closed the site and opened the ssh console to see what's going on: valetudo's RAM usage was average (around 50MB), but CPU usage was a bit weird, being around >20% while the robot was just standing on the dock and doing nothing. In /var/log/upstart/valetudo.log there was nothing wrong also, it was just said that valetudo had started and the robot was connected, nothing else.

I opened the site again and observed for around 5-7 minutes - nothing changed, status was still unknown. Then I issued the command "sudo restart valetudo" to the ssh, it restarted and I immediately got all the required values on the "home" tab. The map was dropped to the default "no map data" state though, but this was expected and was restored soon.

So, any ideas on how to debug this? Anyone encountered similar issues? Seems valetudo got caught somehow in an undefined state and couldn't automatically recover from this, until I simply restarted its process.

Maybe it starts too quick before the firmware is ready to answer its requests? Shouldn't we add some timeout before valetudo process actually starts for the first time after the device (re)boot, or something like that?

rand256, 2019-06-09:
Seems it was my own mistake, sorry.

rand256, 2019-06-10:
At first I erroneously wrote the specific issue not directly connected to original Valetudo, but now I've rewritten it in more general form.

Hypfer, 2019-06-11:
I'm guessing that the msgID is the culprit here. The whole logic around that isn't very nice anyways.

You could try adding logging to each increment of it and see what happens

Hypfer, 2020-05-01:
Is this still an issue?

rand256, 2020-05-01:
Nope.

closed 2020-05-01 by rand256