Hypfer/Valetudo #262: valetudo crash on gen 1

Micha-Btz, 2019-06-11:
Hi there,

since latest release 0.3.1 i get regularly crashes on valetudo. If I should gess, I would mean it's dedicated that my home wlan is not available. If wlan is all time available I didn't expect this.
Here is the log:
`
root@rockrobo:/var/log/upstart# cat valetudo.log 
Loading configuration file: /mnt/data/valetudo/config.json
Dummycloud is spoofing 203.0.113.1:8053 on 127.0.0.1:8053
Webserver running on port 80
Loading configuration file: /mnt/data/valetudo/config.json
Dummycloud is spoofing 203.0.113.1:8053 on 127.0.0.1:8053
Webserver running on port 80
Robot connected
Robot connected
Robot connected
/snapshot/Valetudo/lib/miio/Vacuum.js:80
            throw new Error("Unable to reach vacuum");
            ^

Error: Unable to reach vacuum
    at Vacuum.sendMessage (/snapshot/Valetudo/lib/miio/Vacuum.js:80:19)
    at Vacuum.<anonymous> (/snapshot/Valetudo/lib/miio/Vacuum.js:115:22)
    at ontimeout (timers.js:498:11)
    at tryOnTimeout (timers.js:323:5)
    at Timer.listOnTimeout (timers.js:290:5)
Loading configuration file: /mnt/data/valetudo/config.json
Dummycloud is spoofing 203.0.113.1:8053 on 127.0.0.1:8053
Webserver running on port 80
Robot connected
Robot connected
Robot connected
Robot connected
Robot connected
Robot connected
Robot connected
Robot connected
No response for message: {"method":"event.consume_material_notify","params":[4],"id":1179579316}
Robot connected
Robot connected
Robot connected
Robot connected
Robot connected
Robot connected
root@rockrobo:/var/log/upstart# service valetudo stop 
stop: Unknown instance: 
`

If you need more infos, please telle which one.

Thanks Micha

Hypfer, 2019-06-13:
I don't see any crashes in that log, apart from the first, which was caused by an out-of-sync Message ID

Micha-Btz, 2019-06-17:
Hm, ok. Maybe it is not a crash, maybe valetudo stops itself. 

For me the result is the same, valetudo starts, when the robo starts and is not available when I try to connect to. Just to be clear, need valetudo an always connected wlan to work?

Since the log doesn't contain more infos as posted, how can I find the root cause?

Thanks.

Hypfer, 2019-06-17:
https://github.com/Hypfer/Valetudo/blob/master/deployment/valetudo.conf#L4

This might have something to do with this constraint

Micha-Btz, 2019-06-21:
you are right, I have changed the start conf to

> root@rockrobo:/etc/init# cat valetudo.conf 
> #!upstart
> description "Valetudo"
> 
> 
> start on runlevel [2345]
> #start on started network-interface INTERFACE=wlan0
> stop on runlevel [06]
> 
> oom score 1000
> exec /usr/local/bin/valetudo
> respawn
> respawn limit 10 90

This worked for me, thanks for the hint.

Micha

closed 2019-06-21 by Micha-Btz