Hypfer/Valetudo #263: inconsistent zone coordinates

anthonyk91, 2019-06-11:
I created some zones for different rooms using the /zone tool, and at first it worked ok and I could send the robot to clean different rooms.  Then at some point the coordinates seem to have changed and it tries to go to the wrong area when instructed to clean a zone.

As an example, I created a zone with coordinates [18240,28764,21029,34077], and now after it stopped working properly I created a new zone for the same room, and it has coordinates [17160,18473,22618,21057].  

At some point the map displayed may have rotated 90 degrees? I don't remember how or when it did this, but I think it was random.  I have it set in lab mode so it preserves the map, and it may have been rebooted once or twice, but I don't think any coordinate changes happened with a reboot.

Is there a way of preserving the maps so the zones can be maintained?  I created the zones by drawing rectangles in the /zone tool and then writing them into the config file or the web interface, which is a bit long to repeat regularly.

FW: 1820
I think the valetudo version is 0.3.1 but not sure how to check, the binary has md5 9f9ac0bc9d24f59d1daa172ddbb519e4

thanks

greenimpala, 2019-06-12:
Did you enable map saving mode? You need to enable 'lab status' then explicitly save the map after a clean.

You can do this using the python library `python-miio` — instructions here https://github.com/Hypfer/Valetudo/issues/44#issuecomment-447295298.

After that, re-add your zones and the co-ords should persist correctly :)

Hypfer, 2019-06-12:
Enabling persistent maps will be possible with valetudo 0.4.0
https://github.com/Hypfer/Valetudo/pull/247

There's also a much nicer editor for zones in 0.4.0
https://github.com/Hypfer/Valetudo/pull/157

anthonyk91, 2019-06-12:
Ok thanks, previously I hadn't explicitly done a save, so i've tried that.  Its a bit tricky to test since the map rotation seemed to occur randomly and infrequently (once?), but i'll define new zones and see how it goes.  Also previously I thought it had maintained the previous map and just rotated it, although maybe that doesn't make any sense and it must have generated a new one as part of a clean.

Looking forward to trying the new version.

ministryofsillywalks, 2019-06-12:
But this will only work with the s50 right?
Or will gen1 also support this?

greenimpala, 2019-06-12:
It's interesting that #247 only sets the lab status and does not call `save_map`. With my S50 I *had* to call both for it to persist as expected. If we think this is correct then I am happy to add this to `0.4`.

posixx, 2019-06-16:
Does 0.4.0 also support nogo zones? So if you want to exclude some parts of the total cleanup space, you can use the new editor to create a zone and mark it as nogo zone. The cleanup would wipe anything other then these zones.

matthiasharrer, 2019-06-25:
@greenimpala that might be correct, I actually only verified that the labmode is indeed reported enabled..

closed 2019-07-15 by Hypfer