Hypfer/Valetudo #265: Is the "flashing itself to stock" problem fixed?

imbaimbaimba, 2019-06-13:
Hi guys,

Anyone know if this problem is fixed yet? Mine keeps doing it. So I decided not to spend more time on flashing it, if this is not fixed. So I will install the Mi App tonight, and flush all my dignity in the toilet :)

balu-, 2019-06-14:
Can anyone briefly explain whats happing. 
(Seems to me my robo is back to stock as well, i don't know why or how)

imbaimbaimba, 2019-06-14:
I reflashed multiple times and tried different firmwares, same result... 
-I flash it, and I cannot get access to it
-I start valetudo manually and the interface starts but is unresponsive for a couple of days
-After a few days, it’s working
-A few days later, it has flashes itself back to stock, or just stopped Valetudo

Valetudo worked great for me before, so I believe it is just broken at the moment?

Maybe it would be good to write on the front page, if it is a known problem, so people don’t spend a lot of time installing etc just now?

rand256, 2019-06-14:
I would be better if you've written which exactly version of valetudo are you trying to use on which exactly version of vacuum.

To me it's working quiet fine on gen2 device. I'm using my own modded build based on original master branch from ~5th of june. Since base files didn't change last weeks, I believe original master branch should work ok too for most gen2 users. Can't guess regarding gen1 devices though.

imbaimbaimba, 2019-06-15:
G2 - latest Valetudo. There has not been a new version of Valetudo for a long time, so it’s the same version I have tried multiple times with different vac firmwares begins it. I just tried all the recommended ones really.

But, I have just gone with stock firmware now and the Mi app, which I have never installed before. To be honest, it seems to work a lot better than it ever has. So I guess that’s the price of privacy ;)

rand256, 2019-06-15:
So you actually tried to use not latest but very outdated Valetudo since there were no precompiled releases for some time. I believe truly latest one should provide better experience.

florian-asche, 2019-06-18:
I did not have this issue until now. But @Hypfer had this problem in an non-internet access able network. So there probably is some restore or reboot to other partition thing in it.

florian-asche, 2019-06-18:
I wrote down some documentation about the partitions here: https://github.com/Hypfer/Valetudo/issues/268. Yeah also needs to be documented somewhere, but wanna just write that down for now. I found a partition that has the same version without root, that could be get activated at some point.

But how we check which partition is active without aceess to the robo?

mar565, 2019-06-18:
Unfoutunaly mine went back to stock too. I hope the problem is solved soon, but until then Im going to try the miApp (goodbye privacy)...

imbaimbaimba, 2019-06-18:
Yeah it sucks to ruin the privacy. But to be honest, my vacuum works a lot better now on stock firmware than it ever did(only ever ran with Valetudo before). So guess it isn’t all bad.

closed 2019-06-18 by Hypfer