Hypfer/Valetudo #266: How to build on docker?

bruvv, 2019-06-17:
I have a macbook but it doesn't understand pkg. So I wanted to run docker but where to begin?

Or perhaps have travic upload to github with pre release or beta attached :) ?
https://docs.travis-ci.com/user/deployment/releases/

Skeletorjus, 2019-06-18:
Hi!

I just made a Dockerfile that does this:

```
FROM ubuntu:14.04

# Build:
# docker build -t valetudo-compiler .

# Copy binary to current directory:
# docker run --rm -it -v $(pwd):/host valetudo-compiler bash -c "cp /Valetudo/valetudo /host && chown $(id -u):$(id -g) /host/valetudo"

RUN apt-get update \
; apt-get -y --force-yes install wget git nodejs npm nodejs-legacy \
; npm config set strict-ssl false \
; npm install -g n \
; n latest \
; git clone http://github.com/hypfer/Valetudo \
; cd Valetudo \
; /usr/local/bin/npm install \
```
Save the quoted text in a file named Dockerfile
Run `docker build -t valetudo-compiler .`
Wait
run `docker run --rm -it -v $(pwd):/host valetudo-compiler bash -c "cp /Valetudo/valetudo /host && chown $(id -u):$(id -g) /host/valetudo"`

You should end up with a new build of valetudo in your current directory.

SSH to your vacuum, stop the valetudo service (`service valetudo stop`), replace /usr/local/bin/valetudo with your new binary and restart valetudo (`service valetudo start`).

florian-asche, 2019-06-18:
Whould be great, to have this in the repo.

bruvv, 2019-06-18:
@Skeletorjus great work! 
@Hypfer please add this so we can help you test new builds :)

bruvv, 2019-06-18:
It does work when you run it manually but it should be able to run on any machine when you upload it to hub.docker.com. Only thing is I cannot get it to work :(

MatttiH, 2019-06-20:
Hello,

it seems that the Dockerfile is missiong the
`/usr/local/bin/npm build `
at the end

Skeletorjus, 2019-06-20:
@MatttiH You are right, managed to mess it up when I copied my Dockerfile. Edited the post.

bruvv, 2019-06-26:
Thanks for the PR ! i changed it around a bit to make it much much smaller will do a new PR for it and close this.

closed 2019-06-26 by bruvv

bruvv, 2019-06-26:
PR made: https://github.com/Hypfer/Valetudo/pull/277
Docker: https://hub.docker.com/r/bruvv/valetudo