Hypfer/Valetudo #267: Display information from /mnt/reserve/rrBkBox.csv

Hypfer, 2019-06-18:
`/mnt/reserve/rrBkBox.csv` contains data that might be interesting to see.

There should be a view which displays the latest of each of those values.

```
Factory_Test
MAC_Address
Serial_Number
First_Startup
Left_Wheel_Stalled_Time
Right_Wheel_Stalled_Time
Side_Brush_Stalled_Time
Main_Brush_Stalled_Time
Work_Time
Start_Clicks
Home_Clicks
Left_Bumper_Clicks
Right_Bumper_Clicks
Left_Drop_Clicks
Right_Drop_Clicks
BinOut_Times
AP_Restart_Times
MCU_Restart_Times
Dockin_Times
Battery_Charge_Time
Battery_Discharge_Time
In_Dock_Time
Main_Brush_Work_Time
Side_Brush_Work_Time
Filter_Work_Time
Charge_Cycle
Boot_Loader_Version
Kernel_Version
Kernel_Build_Time
AP_Version
MCU_Version
LDS_Version
LDS_UID
EMMC_IC
DDR_IC
Battery_IC
Compass_IC
Gyro_IC
Charger_IC
Sensor_Dirty_Time
LDS_Bumper_Clicks
Spot_Clicks
Compass2_IC
Clean_Time
Clean_Area
Clean_Count
Back_Time
Spot_Time
Remote_Time
Goto_Time
```

ElePhontitis, 2019-10-27:
@Hypfer Can you tell how this data is generated? Would be interesting to hook on changes here, but it looks like the values in the file are only generated on certain times and not on the actual event itself. Do you know if there is a way to observe when e.g. a bumper gets "clicked"?

Hypfer, 2020-04-12:
The file seems to be updated on every restart of some relevant process.

It also contains error logs + timestamps:

```
root@rockrobo:~# grep Error /mnt/reserve/rrBkBox.csv
2020-02-19 09:46:25,9999,Error_Code,8
2020-02-19 10:09:39,9999,Error_Code,8
2020-02-19 15:23:38,9999,Error_Code,3
2020-02-20 10:06:43,9999,Error_Code,8
2020-02-24 09:56:25,9999,Error_Code,7
2020-03-08 21:00:30,9999,Error_Code,13
2020-03-19 10:01:34,9999,Error_Code,5
```

Maybe rrBkBox means `Rockrobo Blackbox`? Like the ones inside planes

ElePhontitis, 2020-04-14:
@Hypfer Maybe the information in this file could also be relevant for https://github.com/Hypfer/Valetudo/issues/206 ?

closed 2021-01-09 by Hypfer