Hypfer/Valetudo #271: Stuck robot with no errors

olekenneth, 2019-06-19:
I get this in Home Assistant

[image]

The robot has been stuck for a few hours. 

Valetudo has this state:
[image]

dh-harald, 2019-06-20:
same for me 2times in the last 3 days

olekenneth, 2019-07-10:
Same thing happened again. I'm running 0.3.1, which seems to be the newest

Technikte, 2019-07-18:
Please update to newest version (0.4) and if you using the homeassistant auto-config mqtt then restart ha after the succesfull update.

Hypfer, 2020-05-15:
stale issue

closed 2020-05-15 by Hypfer