Hypfer/Valetudo #276: Timer with Zones

sklerotraficon, 2019-06-24:
Times should (if possible) be configurable to have one or more zones attached to them.

Example: 
- Clean zone labeled "Kitchen" every Monday, Wednesday, Saturday at 09:00
- Clean zone labeled "Hallway", "Bathroom" every day at 08:00

If no zone is attached to the timer configration, the current behavior is triggered.

rand256, 2019-06-26:
Current timers that valetudo uses are native firmware-based timers (read - black box), so they just cannot be programmed for zoned cleaning. All they can is starting a global cleaning at the specified time, and all that valetudo does is just sending the data to setup the date when firmware will trigger the cleaning. But for sure it is possible to create our own software timers and do (almost) whatever we want with them.

Actually, I've implemented[[1]](https://github.com/rand256/valetudo/commit/add0846c455c415e5332392413012d7d07a119a7)[[2]](https://github.com/rand256/valetudo/commit/d15219281d7205c6a8729c7ae8eb6c4f285e455e) scheduled zoned cleaning a few weeks ago in my valetudo fork, but in much simpler way currently. It allows setting up a timer with chosen zone coordinates, which then can't be edited but only switched on/off or deleted. Only zone coordinates are stored, so the timer isn't actually linked to the pre-saved zones. Therefore if the zone is edited or even removed then timer settings would nevertheless remain intact. I've chosen this way because of simplicity and the fact that otherwise it will become a mess to count zones (since zoned cleaning allows up to 5 zones max and inside each pre-saved zone you may have more than one sub-zone).

TheMicroDevil, 2019-06-26:
OMG! WOW! 
I had to do this with bash, cron and no gui till now, because there were no such function in valetudo. 
But now....... 
Sorry, got to go and do more tests with it :)
Thank you very much for your [w|f]ork.

Hypfer, 2019-07-15:
Personally, I'm using Home Assistant instead of timers because it's far more flexible in terms of conditions etc.

e.g. "Clean the kitchen when everyone has left the House but not more than once every 24 Hours on Workdays between 9:00 and 17:00"

Of course we could also have timer-based zoned cleanups in valetudo but the HA approach is much "smarter" so to speak.

closed 2020-05-15 by Hypfer