Hypfer/Valetudo #279: Xiaomi Cloud

Trrw, 2019-06-28:
Hello, can I use Valetudo and the Xiaomi Cloud (App) together? How can I activate the cloud? Thanks

Hypfer, 2019-06-28:
"Valetudo - Free your vacuum from the cloud"

closed 2019-06-28 by Hypfer

therealmikz, 2019-09-13:
If you don't mind I'd like to ask you a question about cloud communication from time to time, it could save me some time with reverse engineering it for myself.
In fact I've got one now: I saw you've created two DNAT rules in iptables, I'm talking about these:
`iptables  -t nat -A OUTPUT -p tcp --dport 80   -d 203.0.113.1 -j DNAT --to-destination 127.0.0.1:8053
iptables  -t nat -A OUTPUT -p udp --dport 8053 -d 203.0.113.1 -j DNAT --to-destination 127.0.0.1:8053`
Why did you redirect communication from two different ports to single 8053 one? Does vacuum software communicate with server on two different ports with different protocols (80 and 8053)?

To clarify, I'm going to change these rules (f.ex. 80 -> 8080, 8053 -> 8053) and modify Dummycloud.js to open two ports instead of one using single handler (the one you wrote) and before handler to forward communication to official cloud based on its real IP address resolved by DNS query. I wonder if it's necessary to open the additional port, and if it is, then why.

Hypfer, 2019-09-13:
Honestly I have no idea. I just took that from the initial dummycloud implementation.
It might indeed be unnecessary.

therealmikz, 2019-09-13:
Thank you very much! I'll try to check it for myself then, I hope that I'll find out that's this 8053 is needless and there's no reason to listen on two different ports.
By the way: who's the author of the initial dummycloud? Is there a repo with it?

Hypfer, 2019-09-13:
Nah 8053 is needed. Its port 80 which isn't.

https://github.com/dgiese/dustcloud/tree/master/dummycloud

therealmikz, 2019-09-13:
Thanks again!
I'll try to play with it over the weekend, when I'm finished (it may take some time) I'll provide a link to my fork in this issue for those who still want to use official application :). I assume you don't want to include this feature in your repository (so no PRs) or maybe do you?

Hypfer, 2019-09-13:
No.

henfri, 2019-09-28:
@Hypfer: I was about to open exactly the same Issue and request.
Let me state my usecase:
I want to use Valetudo especially for Mqtt. I am convinced that after a while, we will be using Valetudo only.
But I also know, that I will be in deep trouble, if I switch off the App right now. Thus, I see the usecase of having a switch in the Valetudo webinterface to switch on/off the cloud communication.

What is the reason you don't want to include this feature?

Regards,
Hendrik