Hypfer/Valetudo #285: New release or idiot-proof build manual needed

momo42, 2019-07-11:
Hi there,
I'd really like to upgrade to a slightly newer Valetudo release as 0.3.1 triggers some bugs here.
The thing is, I'm not a developer and unfortunately I wasn't able to find a build guide that is understandable for me.

@Hypfer Would you please put a new release up for people that are not able to "compile" it on their own? *wink* Alternatively an idiot-proof build manual (on Linux) would also do the job for me. I found some docker container but this docker stuff is again another technology I have no idea how to use ...

So, please have a heart for us dummies! :-)

Hypfer, 2019-07-11:
Paid support starts at 270€/h

Hypfer, 2019-07-11:
To explain this publicly for once:

There will be no release until I've been able to get everything in a working and sorta-reliable state as well as test it for a few days.
A release must always contain 0 known problems (as well as hopefully 0 problems at all) so I will not release anything in this state because there are in fact known problems.
Problems which aren't getting solved because fixing these takes time and requires motivation. I am in lack of both. Motivation however is the main concern here. 

This severe lack of motivation is caused by:

- This issue-tracker containing ~80% Invalid "Bugs"
- People constantly nagging me to do anything
- The general abysmal state of the codebase

Valetudo was never meant to be used by end users. Especially not in this state. If it were, it would certainly cost money since support requires _a lot_ of resources.

Valetudo is meant for developers and developers only. If you're not a developer feel free to use this software but _please_ stop contacting me support requests because you won't get any help while also making me hate this project a bit more each time.

Send Pull Requests. Pay people to send Pull Requests for you. I don't care.
I know it's hard to figure out stuff. I know that you'd just like to use this software. I just can't help you.

Btw: This is more of a general statement so please don't take it personally.

momo42, 2019-07-11:
No problem. Thanks for the clear (but friendly) words. I completely understand your points.

I have to say, that up to now I didn't realize that Valetudo is not meant for end users. In fact your install instructions are very straight forward and all went well without errors in the first try. Thanks for that!

From my point as end user I'd be absolutely fine with an open source beta software that contains known bugs. Just put it in the release notes so that users know what a trade-off they're taking.
Even every new Ubuntu release has long bug release notes.

I'm not able to judge the details but I feel that a newer release could be less buggy than the previous you should put it up as long there is no mayor deal breaker in it. This is your project and so you set the rules (and I'm happy with what I already got) but I'm a bit afraid that we'll never see a new release because the state of 0 bugs seems to be out of reach. However, I would be willing to replace annoying bugs with new bugs that are less annoying. ;-)

Heads up!

Hypfer, 2019-07-14:
Closing due to https://github.com/Hypfer/Valetudo/releases/tag/0.4.0

Thx for the the hint regarding a known issues section!

closed 2019-07-14 by Hypfer