Hypfer/Valetudo #291: MQTT support for iobroker

Leviathan09, 2019-07-17:
I would like to implement the MQTT functionality in my ioBroker
I already have a MQTT Server running and Valetduo is connected to it.
But i don't get it how to send commands to the robot to clean a zone.
From the HomeAssistant Insturctions i don't get it
Some ideas/hints how to do that?

Torsten-, 2019-07-18:
Hi,
I figured it out with a look into the code.

Your Payload has to be JSON: {"command":"zoned_cleanup","zone_ids": ["kueche"]}
Published to: valetudo/.../custom_command

Leviathan09, 2019-07-19:
Yes, thank you 👍
Also got the same info yesterday from a guy on roboter-forum.com
Now i know what i need to publish and can go on with my implementation.

closed 2019-07-19 by Leviathan09