Hypfer/Valetudo #293: Issue with Zone Cleaning via Home Assistant

bacumings, 2019-07-20:
**Describe the bug**
Unable to start a zone cleaning from Home Assistant. I do not get any error messages, the vacuum just doesn't begin cleaning. I tried submitting a start_cleaning_zone api call using Postman and get back the message "Error: Zones must be array of at most 5 zones." despite only including one zone in the request.

**To Reproduce**
1. Install Valetudo and setup Home Assistant autodiscovery
2. Create script for zone cleanup per the [Wiki](https://github.com/Hypfer/Valetudo/wiki/Home-Assistant-Integration#example-scriptsyaml-snippet-in-home-assistant-for-zoned-cleaning)
3. Execute script from Home Assistant
4. No error message displayed, but vacuum doesn't begin cleaning.

**Expected behavior**
Vacuum should begin the zoned cleanup

**Screenshots**
None

**Vacuum Model:** 
Gen2

**Valetudo Version:** 
0.4.0

**User-Agent** 
Home Assistant/MQTT
PostmanRuntime/7.15.0

**Additional context**
I was able to use the go_to command via Home Assistant successfully. This is mentioned directly beneath the section linked in the Wiki above.

bacumings, 2019-07-20:
I was able to do this successfully by modifying the format of the script in Home Assistant; the quotation marks need to be done differently than what is indicated in the Wiki. 

Closing and updating Wiki accordingly.

closed 2019-07-20 by bacumings

murmelbahn, 2019-08-22:
I have the exact same problem, please can you post your solution. I used like in the wiki suggested this:

~~~yaml
  vacuum_ash_kitchen:
     alias: "Clean Kitchen"
     sequence:
       - service: vacuum.send_command
         data:
           entity_id: 'vacuum.xiaomi_vacuum_cleaner'
           command: 'zoned_cleanup'
           params:
             'zone_ids': ['Kitchen']
~~~