Hypfer/Valetudo #295: Map Not Loading at all on Safari in MacOS or any browser on iOS

SpecialK1417, 2019-07-23:
Maps will not load on iOS devices tested using safari, firefox, and chrome. Maps load correctly using firefox and chrome on both Win10 and MacOS via local and remote access using nginx reverse proxy. Proxy config was pulled straight from the wiki. All other tabs seem to work correctly on iOS as well.

**To Reproduce:**

Steps to reproduce the behavior:
1. Navigate to site on iOS device.
2. Map tab will just sit and continuously load.
3. Private tabs = same result.

**Expected behavior**

**Screenshots**
iOS Screenshot: 
[image]

Working Computer Screenshot: 
[image]

**Vacuum Model:** Gen2/S5 Firmware Build: 001782 (2 identical vacuums / same issue)

**Valetudo Version:** 0.4.0 

**User-Agent**:
Non-Working: AppleWebKit/Safari
Working Correctly: Firefox/Chrome (Win10 & MacOS)

**Additional context**

leonardpitzu, 2019-07-26:
clear the safari cache on your ios device - i have 2 iphones (ios 12 and ios 13 beta) and one mac (mojave) and on all of them everything is ok on safari.

SpecialK1417, 2019-07-30:
This did not fix the issue. iOS 12.1.2 and mac (high sierra). I've since integrated the Valetudo PNG companion app into home assistant and no longer need to access the site directly from my phone. Thanks!

closed 2019-07-30 by SpecialK1417