Hypfer/Valetudo #296: Can't start zone cleanup

Schattenruf, 2019-07-28:
Hey :)

I have a problem with my robo.
I flashed the package 1782 with the rrlog-patch option and valetudo 0.4.0. But I cannot get my robby to clean just the defined zones. It only states me: There was an error: 500
[image]

Here is what I stated (token anonymized):

`
root@iobroker:/roborock/image# sudo ../dustcloud/devices/xiaomi.vacuum/firmwarebuilder/imagebuilder.sh --firmware=../firmware/v11_001782.fullos.pkg --public-key=../ssh_key/id_rsa.pub --valetudo-path=../valetudo --disable-firmware-updates --ntpserver=192.168.1.1 --rrlogd-patcher=../patcher.py --disable-xiaomi
Unknown Option --disable-xiaomi. Exit.
root@iobroker:/roborock/image# sudo ../dustcloud/devices/xiaomi.vacuum/firmwarebuilder/imagebuilder.sh --firmware=../firmware/v11_001782.fullos.pkg --public-key=../ssh_key/id_rsa.pub --valetudo-path=../valetudo --disable-firmware-updates --ntpserver=192.168.1.1 --rrlogd-patcher=../patcher.py
Script path: ../dustcloud/devices/xiaomi.vacuum/firmwarebuilder
Generate SSH Host Keys if necessary
Decrypt firmware
Unpack firmware
/roborock/image/fw.0S6KqZ/fw /roborock/image
/roborock/image
Replace ssh host keys
Disable SSH firewall rule
Add SSH authorized_keys
reconfiguring network traffic to xiaomi
Creating backup of rrlogd
Trying to patch rrlogd
/roborock/image/fw.0S6KqZ /roborock/image
Original 000108EB:  F2, 04, 03, FF, F7, 4B, FF
Patched  000108EB:  F2, 04, 03, E3, 20, 4B, FF
Original 000106CE:  33, 46, 4B, A8, 10, 22
Patched  000106CE:  33, 46, 47, A8, 10, 22
Patch appears to be successful. Writing to outfile
The patched rrlogd has been written to rrlogd_patch
/roborock/image
Successfully patched rrlogd
Installing valetudo
Pack new firmware
/roborock/image/fw.0S6KqZ/fw /roborock/image
Encrypt firmware
/roborock/image
Copy firmware to output/v11_001782.fullos.pkg and creating checksums
Cleaning up
FINISHED
fb264ae45777b01712b3f3196429236c  output/v11_001782.fullos.pkg
`
After the building -> flashing:
`(venv) root@iobroker:/roborock# mirobo --ip 192.168.1.176 --token xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx update-firmware image/output/v11_001782.fullos.pkg
Going to update from image/output/v11_001782.fullos.pkg
INFO:miio.updater:Serving on 0.0.0.0:42801, timeout 10
INFO:miio.updater:Using local image/output/v11_001782.fullos.pkg (md5: fb264ae45777b01712b3f3196429236c)
Hosting file at http://192.168.1.204:42801/v11_001782.fullos.pkg
Update started!
  0%|                                                                                    | 0/100 [00:00<?, ?it/s]192.168.1.176 - - [28/Jul/2019 13:20:16] "GET /v11_001782.fullos.pkg HTTP/1.1" 200 -
Downloading:  95%|██████████████████████████████████████████████████████████▉   | 95/100 [00:56<00:02,  1.80it/s]INFO:miio.updater:Got a request, should be downloading now.
Downloading:  95%|██████████████████████████████████████████████████████████▉   | 95/100 [00:57<00:02,  1.80it/s]Installation started, please wait until the vacuum reboots
`

I searched the issues but just found the rrlog-patcher thingy. I tried it manually but no chance either.

Can you help me please?

Thank you!

vincegio, 2019-07-28:
Might be a stupid question, but did you reboot the robot manually? 

Flashed Valetudo myself yesterday and it was not working until I rebooted, think it said this in the installation steps or faq.

Schattenruf, 2019-07-28:
yep I did. Valetudo works, just the zone cleanup does not.

Hypfer, 2019-07-29:
To see the error message, open the browser devtools using f12 and take a look at the console.

Does your zone consist of more than 5 zones? The firmware only supports cleaning of up to 5 zones simultaneously.
It's currently possible to create Zones with more than 5 Zones in the GUI. They just won't work.

closed 2020-02-21 by Schattenruf