Hypfer/Valetudo #297: Changing port of valetudo gives no map data

DavidB81, 2019-07-28:
To access my Roborock S5 from outside my home I changed the port of valetudo but the map is not displayed. The message no map data is displayed. 
Port 80 is already used on my router.
I used the command 'echo "env VAC_WEBPORT=8080" >> /etc/init/valetudo.conf && service valetudo restart' to change port number but no map display.

What's the problem and how to be able to access valetudo with an outside access?

Regards,

**Vacuum Model:** Gen2/S5/

**Valetudo Version: ** 0.4.0

rand256, 2019-07-28:
You may want to try adding port 8080 to "map_upload_host" value in config.json. If it works then probably it should be done automatically, but currently it isn't. Maybe just no one tried that. Since at large there's no reason to specify different port on the device itself.

If you need outside access, just setup your router to do port redirection of any external port to port 80 at vacuum's IP, why do you need the vacuum itself to listen on another port?

DavidB81, 2019-07-29:
Actually I did not redirect the port to port 80 of my Roborock.
With the right configuration of the router the external access works.
Thank you for opening my eyes.
Regards

closed 2019-07-29 by DavidB81