Hypfer/Valetudo #303: No Map Data

marciogranzotto, 2019-08-06:
**No Map Data**

Ok, so I rooted my Mi Robot Gen 1 using dustcloud's tutorial and v11_003468 (created image). Follow [this tutorial](https://community.home-assistant.io/t/add-map-to-xiaomi-vacuum-cleaner/33155/119?u=marciogranzotto) to install Valetudo.
Everything worked except for the maps. Only got the "No Map Data" map.

After that I decided to start over, I used [Dennis's Dustbuilder](https://dustbuilder.xvm.mit.edu) to create a image of v11_003452 with Valetudo v0.4.0.
Same results.

I can control my vaccum with the web UI, Xiaomi App and Flolevac app, but none of them show the map.
I also did several full cleans, rebooted the vaccum with the command line and turning off/on with the device's button.

My `/mnt/data/rockrobo/map/` folder looks empty, but my `last_maps` file has something in it
```shell
root@rockrobo:~# ls -la /mnt/data/rockrobo/map/
total 8
drwxr-xr-x 2 root root 4096 Dec 31  1969 .
drwxr-xr-x 5 root root 4096 Aug  6 01:45 ..
root@rockrobo:~# ls -la /mnt/data/rockrobo/
total 564
drwxr-xr-x  5 root root   4096 Aug  6 01:45 .
drwxr-xr-x  8 root root   4096 Aug  4 04:04 ..
-rw-r--r--  1 root root     40 Aug  6 01:38 ChargerPos.data
-rw-r--r--  1 root root    370 Aug  6 01:45 RoboController.cfg
-rw-r--r--  1 root root     40 Aug  6 01:38 StartPos.data
-rw-r--r--  1 root root    282 Aug  4 04:03 Update.pkg.inf
-rw-r--r--  1 root root     43 Aug  6 01:45 VacuumAndBrush.cfg
-rw-r--r--  1 root root  92268 Aug  6 01:43 appproxy.map
-rw-r--r--  1 root root    149 Aug  4 04:40 apptimer.cfg
-rw-r--r--  1 root root      1 Aug  4 02:36 lab.cfg
-rw-r--r--  1 root root 365640 Aug  6 01:43 last_map
-rw-r--r--  1 root root     44 Aug  5 08:30 lds_info.txt
-rw-r--r--  1 root root   2001 Aug  4 01:41 localtime
-rw-r--r--  1 root root   2001 Aug  4 01:41 localtime.bak
drwxr-xr-x  2 root root   4096 Dec 31  1969 map
-rw-r--r--  1 root root  22528 Aug  6 01:19 robot.db
drwxr-xr-x 42 root root  16384 Aug  6 01:48 rrlog
-rw-r--r--  1 root root     81 Aug  6 01:43 slam_info.cfg
drwxr-xr-x  2 root root   4096 Aug  4 04:28 sounds
-rw-r--r--  1 root root     17 Aug  4 01:41 timezone
-rw-r--r--  1 root root     10 Aug  3 23:22 wlanmgr.conf
root@rockrobo:~#
```

**Screenshots**

[image]

**Vacuum Model:** Xiaomi Mi Robot Gen1
**Valetudo Version: ** 0.4.0

**User-Agent** Any

**Extras**
```shell
root@rockrobo:/mnt/data/valetudo# cat /etc/rc.local
#!/bin/sh -e
#
# rc.local
#
# This script is executed at the end of each multiuser runlevel.
# Make sure that the script will "" on success or any other
# value on error.
#
# In order to enable or disable this script just change the execution
# bits.
#
# By default this script does nothing.

###########################################################
### VALETUDO RC.LOCAL
###########################################################
#
# Add to /etc/rc.local before the "exit 0" command

iptables         -F OUTPUT
ip6tables        -F OUTPUT
iptables  -t nat -F OUTPUT
iptables  -t nat -A OUTPUT -p tcp --dport 80   -d 203.0.113.1 -j DNAT --to-destination 127.0.0.1:8053
iptables  -t nat -A OUTPUT -p udp --dport 8053 -d 203.0.113.1 -j DNAT --to-destination 127.0.0.1:8053
iptables         -A OUTPUT                     -d 203.0.113.1/32  -j REJECT
ip6tables        -A OUTPUT                     -d 2001:db8::1/128 -j REJECT

exit 0

root@rockrobo:/mnt/data/valetudo# cat /etc/hosts
127.0.0.1       localhost
127.0.1.1       rockrobo

::1             localhost ip6-localhost ip6-loopback
ff02::1         ip6-allnodes
ff02::2         ip6-allrouters

###########################################################
### VALETUDO HOSTS
###########################################################
#
# Add to /etc/hosts
#

203.0.113.1  ot.io.mi.com
2001:db8::1  ot.io.mi.com
203.0.113.1  ott.io.mi.com
2001:db8::1  ott.io.mi.com
203.0.113.1  us.ot.io.mi.com
2001:db8::1  us.ot.io.mi.com
203.0.113.1  us.ott.io.mi.com
2001:db8::1  us.ott.io.mi.com
203.0.113.1  sg.ot.io.mi.com
2001:db8::1  sg.ot.io.mi.com
203.0.113.1  sg.ott.io.mi.com
2001:db8::1  sg.ott.io.mi.com
203.0.113.1  st.ot.io.mi.com
2001:db8::1  st.ot.io.mi.com
203.0.113.1  st.ott.io.mi.com
2001:db8::1  st.ott.io.mi.com
203.0.113.1  de.ot.io.mi.com
2001:db8::1  de.ot.io.mi.com
203.0.113.1  de.ott.io.mi.com
2001:db8::1  de.ott.io.mi.com
203.0.113.1  ru.ot.io.mi.com
2001:db8::1  ru.ot.io.mi.com
203.0.113.1  ru.ott.io.mi.com
2001:db8::1  ru.ott.io.mi.com
203.0.113.1  in.ot.io.mi.com
2001:db8::1  in.ot.io.mi.com
203.0.113.1  in.ott.io.mi.com
2001:db8::1  in.ott.io.mi.com
203.0.113.1  tw.ot.io.mi.com
2001:db8::1  tw.ot.io.mi.com
203.0.113.1  tw.ott.io.mi.com
2001:db8::1  tw.ott.io.mi.com
203.0.113.1  ea.ot.io.mi.com
2001:db8::1  ea.ot.io.mi.com
203.0.113.1  ea.ott.io.mi.com
2001:db8::1  ea.ott.io.mi.com
203.0.113.1  pv.ot.io.mi.com
2001:db8::1  pv.ot.io.mi.com
203.0.113.1  pv.ott.io.mi.com
2001:db8::1  pv.ott.io.mi.com
203.0.113.1  cdn.cnbj0.files.fds.api.xiaomi.com
2001:db8::1  cdn.cnbj0.files.fds.api.xiaomi.com
203.0.113.1  cnbj0.fds.api.xiaomi.com
2001:db8::1  cnbj0.fds.api.xiaomi.com
203.0.113.1  cnbj2.fds.api.xiaomi.com
2001:db8::1  cnbj2.fds.api.xiaomi.com
203.0.113.1  awsbj0.fds.api.xiaomi.com
2001:db8::1  awsbj0.fds.api.xiaomi.com
203.0.113.1  awsbj0-files.fds.api.xiaomi.com
2001:db8::1  awsbj0-files.fds.api.xiaomi.com
203.0.113.1  awsusor0.fds.api.xiaomi.com
2001:db8::1  awsusor0.fds.api.xiaomi.com
203.0.113.1  awsusor0-files.fds.api.xiaomi.com
2001:db8::1  awsusor0-files.fds.api.xiaomi.com
203.0.113.1  awssgp0.fds.api.xiaomi.com
2001:db8::1  awssgp0.fds.api.xiaomi.com
203.0.113.1  awssgp0-files.fds.api.xiaomi.com
2001:db8::1  awssgp0-files.fds.api.xiaomi.com
203.0.113.1  awsde0.fds.api.xiaomi.com
2001:db8::1  awsde0.fds.api.xiaomi.com
203.0.113.1  awsde0-files.fds.api.xiaomi.com
2001:db8::1  awsde0-files.fds.api.xiaomi.com
203.0.113.1  ksyru0-eco.fds.api.xiaomi.com
2001:db8::1  ksyru0-eco.fds.api.xiaomi.com
203.0.113.1  ksyru0-eco-files.fds.api.xiaomi.com
2001:db8::1  ksyru0-eco-files.fds.api.xiaomi.com
203.0.113.1  awsind0-eco.fds.api.xiaomi.com
2001:db8::1  awsind0-eco.fds.api.xiaomi.com
203.0.113.1  awsind0-eco-files.fds.api.xiaomi.com
2001:db8::1  awsind0-eco-files.fds.api.xiaomi.com

root@rockrobo:/mnt/data/valetudo# cat config.json
{
  "spots": [],
  "areas": [],
  "mqtt": {
    "enabled": true,
    "identifier": "rockrobo",
    "topicPrefix": "valetudo",
    "autoconfPrefix": "homeassistant",
    "broker_url": "##REDACTED##",
    "provideMapData": true,
    "caPath": ""
  },
  "dummycloud": {
    "spoofedIP": "203.0.113.1",
    "bindIP": "127.0.0.1"
  },
  "httpAuth": {
    "enabled": false,
    "username": "valetudo",
    "password": "valetudo"
  },
  "allowSSHKeyUpload": true,
  "map_upload_host": "http://192.168.1.133"
}
```

marciogranzotto, 2019-08-06:
Github Issues are really helpful to make us understand what we screwed up after opening it.
For future reference, the problem was that I changed `mnt/data/valetudo/config.json` `map_upload_host` parameter to point out to my MQTT broker, but it should not be changed from `127.0.0.1`.
Maybe documenting what that parameter does would be helpful though.

Sorry for the issue!

closed 2019-08-06 by marciogranzotto

marvinwankersteen, 2019-10-13:
I have exactly the same problem with valetudo 0.4.0 and the image `v11_003468.fullos.pkg` built with dustclouds firmwareimagebuilder.
In `/mnt/data/valetudo/config.json` is `"map_upload_host": "http://127.0.0.1"` already set.

In the browsers developer tools I can see, that the robot send the positions constantly the same points of the "NO MAP DATA" to the browser via websocket. So I think the problem must be on the robot.

How can I further debug that problem? I have really no idea why this is happening.

HorizonKane, 2019-10-25:
Same problem here :(

HorizonKane, 2019-10-26:
Okay it's working now.

Solution: 
1) Got o bed
2) sleep
3) get up
4) Do nothing but pressing start like yesterday evening
5) Map suddenly works *magic*

onggie, 2019-11-27:
@marvinwankersteen did you find a solution to this problem?

RoStief, 2020-01-15:
Hello, i had the same problem, 
i have set up the image on windows, therefore the rc.local file had the wrong encoding.
So if you are running into the problem:
- install dos2unix on the robot:
   sudo apt-get install dos2unix
- run dos2unix /etc/rc.local
- reboot